joan2937/pigpio #435: hardwareclock() on the pi 4  outputs wrong frequency

JakeQuinn77, 2021-03-04:
I have a problem setting the hardwareclock() on the pi 4 

when  `gpioHardwareClock(6,5000000);` is excuted the output frequency is way below than the expected 5 Mhz. 

[Logic analyzer image.](https://i.stack.imgur.com/Hx4QM.png)

Aint the second parameter the expected output frequency? I also tried playing with different values and the result is kind of the same. I have another code for setting the clock on CLK2(GPIO6) and it can spit out the correct frequency so i know the pin is not destroyed and is indeed capable of outputting a clock

How i use the library is i include `#include <pigpio.h>`  on my C file then compile with `-lpigpio`. 
My pigpiod version is 71
Kernel version: Linux raspberrypi 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux

guymcswain, 2021-03-04:
> My pigpiod version is 71

Please upgrade to the latest version.  Pi4 support came after v71.
```
wget https://github.com/joan2937/pigpio/archive/master.zip
unzip master.zip
cd pigpio-master
make
sudo make install
```

JakeQuinn77, 2021-03-04:
I have update my pigpio and now is version `79` . Unfortunately that did not fix the problem. I made another file to make sure none of my code was affecting it.

```
#include <pigpio.h>

int main(){

    gpioInitialise();
    gpioHardwareClock(6,5000000);
}
```
I compiled it with `gcc -o main main.c -lpigpio`

the result are the same :

[image]

guymcswain, 2021-03-04:
Try running the test `sudo ./x_pigpio` from the install directory.  It will use gpio 25 so make sure nothing is connected there.

JakeQuinn77, 2021-03-04:
So far eveything is a pass.

```
Testing pigpio C I/F
pigpio version 79.
Hardware revision 12595473.
Mode/PUD/read/write tests.
TEST  1.1  PASS (set mode, get mode: 0)
TEST  1.2  PASS (set pull up down, read: 1)
TEST  1.3  PASS (set pull up down, read: 0)
TEST  1.4  PASS (write, get mode: 1)
TEST  1.5  PASS (read: 0)
TEST  1.6  PASS (write, read: 1)
PWM dutycycle/range/frequency tests.
TEST  2.1  PASS (set PWM range, set/get PWM frequency: 10)
TEST  2.2  PASS (get PWM dutycycle: 0)
TEST  2.3  PASS (set PWM dutycycle, callback: 0)
TEST  2.4  PASS (get PWM dutycycle: 128)
TEST  2.5  PASS (set PWM dutycycle, callback: 40)
TEST  2.6  PASS (set/get PWM frequency: 100)
TEST  2.7  PASS (callback: 400)
TEST  2.8  PASS (set/get PWM frequency: 1000)
TEST  2.9  PASS (callback: 4000)
TEST  2.10 PASS (get PWM range: 255)
TEST  2.11 PASS (get PWM real range: 200)
TEST  2.12 PASS (set/get PWM range: 2000)
TEST  2.13 PASS (get PWM real range: 200)
PWM/Servo pulse accuracy tests.
TEST  3.1  PASS (get servo pulsewidth: 500)
TEST  3.2  PASS (set servo pulsewidth: 40000)
TEST  3.3  PASS (get servo pulsewidth: 1500)
TEST  3.4  PASS (set servo pulsewidth: 13333)
TEST  3.5  PASS (get servo pulsewidth: 2500)
TEST  3.6  PASS (set servo pulsewidth: 8000)
TEST  3.7  PASS (set/get PWM frequency: 1000)
TEST  3.8  PASS (set PWM range: 200)
TEST  3.9  PASS (get PWM dutycycle: 20)
TEST  3.10 PASS (set PWM dutycycle: 200)
TEST  3.11 PASS (get PWM dutycycle: 40)
TEST  3.12 PASS (set PWM dutycycle: 400)
TEST  3.13 PASS (get PWM dutycycle: 60)
TEST  3.14 PASS (set PWM dutycycle: 600)
TEST  3.15 PASS (get PWM dutycycle: 80)
TEST  3.16 PASS (set PWM dutycycle: 800)
Pipe notification tests.
TEST  4.1  PASS (notify open/begin: 0)
TEST  4.2  PASS (notify pause: 0)
TEST  4.3  PASS (notify close: 0)
TEST  4.4  PASS (sequence numbers ok: 1)
TEST  4.5  PASS (gpio toggled ok: 1)
TEST  4.6  PASS (number of notifications: 80)
Waveforms & serial read/write tests.
TEST  5.1  PASS (callback, set mode, wave clear: 0)
TEST  5.2  PASS (pulse, wave add generic: 4)
TEST  5.3  PASS (wave tx repeat: 9)
TEST  5.4  PASS (callback: 50)
TEST  5.5  PASS (wave tx stop: 0)
TEST  5.6  PASS (serial read open: 0)
TEST  5.7  PASS (wave clear, wave add serial: 3405)
TEST  5.8  PASS (wave tx start: 6811)
TEST  5.9  PASS (NOT APPLICABLE: 0)
TEST  5.10 PASS (NOT APPLICABLE: 0)
TEST  5.11 PASS (wave tx busy, serial read: 0)
TEST  5.12 PASS (serial read close: 0)
TEST  5.13 PASS (wave get micros: 6158148)
TEST  5.14 PASS (wave get high micros: 6158148)
TEST  5.15 PASS (wave get max micros: 1800000000)
TEST  5.16 PASS (wave get pulses: 3405)
TEST  5.17 PASS (wave get high pulses: 3405)
TEST  5.18 PASS (wave get max pulses: 12000)
TEST  5.19 PASS (wave get cbs: 6810)
TEST  5.20 PASS (wave get high cbs: 6810)
TEST  5.21 PASS (wave get max cbs: 25016)
TEST  5.22 PASS (wave create pad, count==1, wid==: 0)
TEST  5.23 PASS (wave create pad, count==2, wid==: 1)
TEST  5.24 PASS (delete wid==0 success: 0)
TEST  5.25 PASS (No more CBs using wave create: -67)
TEST  5.26 PASS (wave create pad, count==3, wid==: 0)
TEST  5.27 PASS (wave chain [1,0]: 0)
TEST  5.28 PASS (callback count==: 5)
Trigger tests
TEST  6.1  PASS (gpio trigger count: 5)
TEST  6.2  PASS (gpio trigger pulse length: 150)
Watchdog tests.
TEST  7.1  PASS (set watchdog on count: 39)
TEST  7.2  PASS (set watchdog off count: 0)
Bank read/write tests.
TEST  8.1  PASS (read bank 1: 0)
TEST  8.2  PASS (read bank 1: 33554432)
TEST  8.3  PASS (clear bank 1: 0)
TEST  8.4  PASS (set bank 1: 1)
TEST  8.5  PASS (read bank 2: 0)
TEST  8.6  PASS (clear bank 2: 0)
TEST  8.7  PASS (NOT APPLICABLE: 0)
TEST  8.8  PASS (set bank 2: 0)
TEST  8.9  PASS (NOT APPLICABLE: 0)
Script store/run/status/stop/delete tests.
TEST  9.1  PASS (store/run script: 100)
TEST  9.2  PASS (run script/script status: 201)
TEST  9.3  PASS (run/stop script/script status: 110)
TEST  9.4  PASS (delete script: 0)
```

guymcswain, 2021-03-04:
Hmm. Everything looks ok there.  I'll do some tests on my pi4 later today.

guymcswain, 2021-03-04:
On my pi400, I've done
`pigs hc 6 5000000`
and I get a 5MHz output on GPIO 6, but it's not stable.

JakeQuinn77, 2021-03-04:
I did a fresh install on my SD card and ran the same code above, its still generating that 393Khz  instead of the 5Mhz. 

I tried to run `pigs hc 6 5000000`  and it still gets me more or less the same frequency. 

What is going on here. pi400 and pi 4 is basically the same.

JakeQuinn77, 2021-03-04:
This is the code that let me output a 5MHz signal on clock 6, this does not use the pigpio library, it is a modified version of joan's [minimal_clk](http://abyz.me.uk/rpi/pigpio/examples.html#Misc_minimal_clk) example (modified to work with pi 4)

```
#include <stdio.h>
#include <stdlib.h>
#include <stdarg.h>
#include <unistd.h>
#include <stdint.h>
#include <string.h>
#include <fcntl.h>
#include <errno.h>
#include <sys/mman.h>
#include <sys/stat.h>
#include <sys/types.h>

static volatile uint32_t piPeriphBase = 0xFE000000;

static volatile int pi_is_2711 = 0;

#define SYST_BASE  (piPeriphBase + 0x003000)
#define DMA_BASE   (piPeriphBase + 0x007000)
#define CLK_BASE   (piPeriphBase + 0x101000)
#define GPIO_BASE  (piPeriphBase + 0x200000)

#define BSCS_BASE  (piPeriphBase + 0x214000)

#define DMA_LEN   0x1000 /* allow access to all channels */
#define CLK_LEN   0xA8
#define GPIO_LEN  0xF4
#define SYST_LEN  0x1C
#define BSCS_LEN  0x40

#define GPSET0 7
#define GPSET1 8

#define GPCLR0 10
#define GPCLR1 11

#define GPLEV0 13
#define GPLEV1 14

#define GPPUD     37
#define GPPUDCLK0 38
#define GPPUDCLK1 39

/* BCM2711 has different pulls */

#define GPPUPPDN0 57
#define GPPUPPDN1 58
#define GPPUPPDN2 59
#define GPPUPPDN3 60

#define SYST_CS  0
#define SYST_CLO 1
#define SYST_CHI 2

static volatile uint32_t  *gpioReg = MAP_FAILED;
static volatile uint32_t  *systReg = MAP_FAILED;
static volatile uint32_t  *bscsReg = MAP_FAILED;

#define PI_BANK (gpio>>5)
#define PI_BIT  (1<<(gpio&0x1F))

#define CLK_GP2_CTL 32
#define CLK_GP2_DIV 33

#define CLK_PASSWD  (0x5A<<24)

#define CLK_CTL_MASH(x)((x)<<9)
#define CLK_CTL_BUSY    (1 <<7)
#define CLK_CTL_KILL    (1 <<5)
#define CLK_CTL_ENAB    (1 <<4)
#define CLK_CTL_SRC(x) ((x)<<0)

#define CLK_DIV_DIVI(x) ((x)<<12)
#define CLK_DIV_DIVF(x) ((x)<< 0)

#define CLK_CTL_SRC_OSC  1  /* 19.2 MHz */
#define CLK_CTL_SRC_PLLC 5  /* 1000 MHz */
#define CLK_CTL_SRC_PLLD 6  /*  500 MHz */
#define CLK_CTL_SRC_HDMI 7  /*  216 MHz */

static volatile uint32_t  *clkReg  = MAP_FAILED;

/* gpio modes. */

#define PI_INPUT  0
#define PI_OUTPUT 1
#define PI_ALT0   4
#define PI_ALT1   5
#define PI_ALT2   6
#define PI_ALT3   7
#define PI_ALT4   3
#define PI_ALT5   2

unsigned gpioHardwareRevision(void)
{
   static unsigned rev = 0;

   FILE *filp;
   char buf[512];
   char term;
   int chars=4; /* number of chars in revision string */

   filp = fopen ("/proc/cpuinfo", "r");

   if (filp != NULL)
   {
      while (fgets(buf, sizeof(buf), filp) != NULL)
      {
         if (!strncasecmp("revision", buf, 8))
         {
            if (sscanf(buf+strlen(buf)-(chars+1),
               "%x%c", &rev, &term) == 2)
            {
               if (term != '\n') rev = 0;
               else rev &= 0xFFFFFF; /* mask out warranty bit */
            }
         }
      }

      fclose(filp);
   }

   if (filp = fopen("/proc/device-tree/soc/ranges" , "rb"))
   {
      if (fread(buf, 1, sizeof(buf), filp) >= 8)
      {
         piPeriphBase = buf[4]<<24 | buf[5]<<16 | buf[6]<<8 | buf[7];
         if (!piPeriphBase)
            piPeriphBase = buf[8]<<24 | buf[9]<<16 | buf[10]<<8 | buf[11];

         if (piPeriphBase == 0xFE00000) pi_is_2711 = 1;
      }
      fclose(filp);
   }

   return rev;
}

static uint32_t * initMapMem(int fd, uint32_t addr, uint32_t len)
{
    return (uint32_t *) mmap(0, len,
       PROT_READ|PROT_WRITE|PROT_EXEC,
       MAP_SHARED|MAP_LOCKED,
       fd, addr);
}

int gpioInitialise(void)
{
   int fd;

   gpioHardwareRevision(); /* sets rev and peripherals base address */

   fd = open("/dev/mem", O_RDWR | O_SYNC) ;

   if (fd < 0)
   {
      fprintf(stderr,
         "This program needs root privileges.  Try using sudo\n");
      return -1;
   }

   gpioReg  = initMapMem(fd, GPIO_BASE,  GPIO_LEN);
   systReg  = initMapMem(fd, SYST_BASE,  SYST_LEN);
   bscsReg  = initMapMem(fd, BSCS_BASE,  BSCS_LEN);
   clkReg = initMapMem(fd, CLK_BASE,  CLK_LEN);

   close(fd);

   if ((gpioReg == MAP_FAILED) ||
       (systReg == MAP_FAILED) ||
       (bscsReg == MAP_FAILED))
   {
      fprintf(stderr,
         "Bad, mmap failed\n");
      return -1;
   }
   return 0;
}

void gpioSetMode(unsigned gpio, unsigned mode)
{
   int reg, shift;

   reg   =  gpio/10;
   shift = (gpio%10) * 3;

   gpioReg[reg] = (gpioReg[reg] & ~(7<<shift)) | (mode<<shift);
}

static int initClock2(int divI, int divF, int MASH)
{
   clkReg[CLK_GP2_CTL] = CLK_PASSWD | CLK_CTL_KILL;

   /* wait for clock to stop */
   while (clkReg[CLK_GP2_CTL] & CLK_CTL_BUSY)
      usleep(10);

   clkReg[CLK_GP2_DIV] =
      (CLK_PASSWD | CLK_DIV_DIVI(divI) | CLK_DIV_DIVF(divF));

   usleep(10);

   clkReg[CLK_GP2_CTL] =
      (CLK_PASSWD | CLK_CTL_MASH(MASH) | CLK_CTL_SRC(CLK_CTL_SRC_OSC));

   usleep(10);

   clkReg[CLK_GP2_CTL] |= (CLK_PASSWD | CLK_CTL_ENAB);
}

static int termClock2()
{
   clkReg[CLK_GP2_CTL] = CLK_PASSWD | CLK_CTL_KILL;

   /* wait for clock to stop */

   while (clkReg[CLK_GP2_CTL] & CLK_CTL_BUSY)
      usleep(10);

}

int main(int argc, char *argv[])
{

    if (gpioInitialise() < 0) return 1;

    initClock2(10,144,0);
    gpioSetMode(6, PI_ALT0);

}
```

guymcswain, 2021-03-04:
> What is going on here. pi400 and pi 4 is basically the same.

That is my understanding.  You're not running other apps that touch the pi hardware such as pi-fm for example?

drakejest, 2021-03-04:
Did you scope your clock to confirm it was really about 5Mhz? 

Do you think you can run the first code i gave and check your output clock for me? I do not have another pi 4 to cross check.

I am suspecting that the function might have not calculated the clock fraction and clock integer.

guymcswain, 2021-03-04:
Yeah, I looked at the output with my scope and it is measuring 5MHz.  It's a bit jumpy when on auto trigger.  I slow it down and the waveform jumps but at a much slower rate.  I've not used the clock output much so I'm not sure if this is expected or something else that I need to look into.

I didn't compile your C code snippet since the pigs hc calls the same function.   Since your systems behaves the same using either pigs or the C code then there is a difference between our systems that remains unknown.  For what its worth, the pigpio website suggest the way to compile is as follows:
```
gcc -Wall -pthread -o prog prog.c -lpigpio -lrt
```

guymcswain, 2021-03-04:
Just thought of this:  If your program is somehow compiled incorrectly and you never terminate by calling gpioTerminate() then pigs will run the same executable.

Try rebooting your system - to kill your program - then run `sudo pigpiod` and then `pigs hc 6 5000000`

drakejest, 2021-03-04:
> Try rebooting your system - to kill your program - then run `sudo pigpiod` and then `pigs hc 6 5000000`

Thank you for taking the time to probe your board, unfortunately fresh from reboot  did not solve the problem it would still output in the kHz range.

drakejest, 2021-03-05:
@guymcswain Oh my! I think i know whats wrongs with the library ! I was CORRECT (i think) that the library is calculating the fractional and interger variables needed for the clock wrong, I traced back [gpioHardwareClock()](https://github.com/joan2937/pigpio/blob/master/pigpio.c#L12965) and basically inside it is just error checking and the only function responsible for setting the clock settings is the [initHWClk()](https://github.com/joan2937/pigpio/blob/master/pigpio.c#L13012) so i tried to trace as much as i can what parameters it is passing, i could only make up this. I am missing DivI

```
//initHWClk(clkCtl, clkDiv, clkSrc, clkDivI, clkDivF, clkMash);
initHWClk(32,33,1,???,0,0); 
```

So i copied  initHWClK and i used the numbers i used for the code that worked for me (which is DivI = 10, DivF=144) 

```
#include <unistd.h>
#include <fcntl.h>
#include <sys/mman.h>

#include <pigpio.h>

static volatile uint32_t piPeriphBase = 0xFE000000;

#define CLK_BASE   (piPeriphBase + 0x101000)
#define CLK_LEN    0xA8

#define CLK_CTL_BUSY    (1 <<7)
#define CLK_CTL_KILL    (1 <<5)
#define CLK_CTL_ENAB    (1 <<4)

#define BCM_PASSWD  (0x5A<<24)

#define CLK_DIV_DIVI(x) ((x)<<12)
#define CLK_DIV_DIVF(x) ((x)<< 0)
#define CLK_CTL_MASH(x) ((x)<<9)
#define CLK_CTL_SRC(x)  ((x)<<0)

static volatile uint32_t * clkReg  = MAP_FAILED;

static uint32_t * initMapMem(int fd, uint32_t addr, uint32_t len)
{
    return (uint32_t *) mmap(0, len,
       PROT_READ|PROT_WRITE|PROT_EXEC,
       MAP_SHARED|MAP_LOCKED,
       fd, addr);
}

static void initHWClk
   (int clkCtl, int clkDiv, int clkSrc, int divI, int divF, int MASH)
{

    // Not in the original function but needed inorder to have access to clkReg register
    int fd = open("/dev/mem", O_RDWR | O_SYNC) ;
    clkReg = initMapMem(fd, CLK_BASE,  CLK_LEN);

   /* kill the clock if busy, anything else isn't reliable */

   if (clkReg[clkCtl] & CLK_CTL_BUSY)
   {
      do
      {
         clkReg[clkCtl] = BCM_PASSWD | CLK_CTL_KILL;
      }
      while (clkReg[clkCtl] & CLK_CTL_BUSY);
   }

   clkReg[clkDiv] = (BCM_PASSWD | CLK_DIV_DIVI(divI) | CLK_DIV_DIVF(divF));

   usleep(10);

   clkReg[clkCtl] = (BCM_PASSWD | CLK_CTL_MASH(MASH) | CLK_CTL_SRC(clkSrc));

   usleep(10);

   clkReg[clkCtl] |= (BCM_PASSWD | CLK_CTL_ENAB);
}

int main(){

    gpioInitialise();

    //initHWClk(clkCtl, clkDiv, clkSrc, clkDivI, clkDivF, clkMash);
    initHWClk(32,33,1,10,144,0);

    gpioSetMode(6,PI_ALT0);

}
```

And lo and behold a 5Mhz (or close) clock signal.

[image]

To further solidify my theory i would like to know what gpiohardwareclock is really passing as parameters, [In this line of code ](https://github.com/joan2937/pigpio/blob/master/pigpio.c#L13024)inside gpioHardwareClock() there is a some sort of debug of all the parameters used, how do i make this show up in my terminal ?

drakejest, 2021-03-05:
@joan2937 @guymcswain so the code responsible for choosing the DIV_I , DIV_F, and mash is [chooseBestClock()](https://github.com/joan2937/pigpio/blame/c33738a320a3e28824af7807edafda440952c05d/pigpio.c#L5161) and i dont why it works in your case but  when i ran chooseBestClock() for 5MHz it gives an output of 

```
chooseBestClock returned valid
DIV: 100
FRAC: 0
CLOCK: 1
FREQUENCY ASKED: 5000000
```

so that means if we follow the formula for calculating the frequency according to the [cpu datasheet](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711/rpi_DATA_2711_1p0.pdf#page=107) the output should be 

[image]

The Units are in megaHertz. I have also included some other values that is a bit closer to 5mHz  and i have tested those values too with the code i posted above, and they do hold TRUE.

guymcswain, 2021-03-05:
I ran your code:
```
#include <pigpio.h>

int main(){

    gpioInitialise();
    gpioHardwareClock(6,5000000);
    gpioTerminate();  // inserted to clean up all threads before exiting
}
```
It produces a 5MHz clock on GPIO6.  That issue is closed.  Is there a different/new issue that you are trying to raise?

drakejest, 2021-03-06:
I would not say its different, but it is related. As what we have been trying to figure out why my pi 4 cannot output the proper clock. I cannot say if it is MY specific pi4 board or other pi4 board have this problem too ( i only have 1 pi4 sadly). literally a sample size of 1

So i looked into the source files how `gpioHardwareClock()` works internally and its basically calling 2 key functions, `chooseBestClock()` which is responsible or calculating the settings and  `initHWClK ` which is reponsible for actually setting the clock using the output numbers of chooseBestClock.

Now what i did is i tested both of those functions individually on `initHWCLK()` i used the formula on the BCM datasheet as guide to calculate the parameters needed by initHWCLK() to produce the proper clock signal. and it works i get the expected frequency on my logic analyzer.

Now here is the problem when i tested chooseBestClock and i have asked it for 5mHz, the numbers it spit out when i plugged it into my spreadsheet it NOT 5 MEGA HERTZ  but 500 KILO HERTZ (see my previous comments) and when i  ran those numbers into initHWCLK() i am seeing about 400-600 KILO HERTZ on the pin. So something is hapenning in choosBestClock() thats causing it  to calaculate wrong number and by extension my gpioHardwareClock() fails too because it is calling chooseBestClock

closed 2021-03-22 by guymcswain