joan2937/pigpio #436: pi1.write(3, 0) causing a timeout crash

rfresh737, 2021-03-08:
I'm new to using your library. I have a Python script that I want to use set_glitch_filter() on for an encoder I'm using to set the heading bug on an XPlane aircraft.

Here is my snippet of code:

`import pigpio
pi1 = pigpio.pi()
pi1.write(3, 0) # set GPIO pin low
#pi1.set_glitch_filter(3, 100)
`
The line pi1.write(3, 0) causes my script to crash back to the command prompt after about 5 seconds. If I comment that line out, the scripts runs fine and does not time out.

Being new, I guess I'm missing setting something so it doesn't timeout so fast?

Thank you...

guymcswain, 2021-03-08:
You will have a conflict if another driver is controlling I2C bus with GPIO 3.

Are you trying to de-glitch the i2c bus?

rfresh737, 2021-03-08:
No, not the entire i2c bus anyway. But, I do have my encoder connected to the MCP23017 which is connected to the i2c bus. So perhaps that is the conflict? I'll try connecting my encoder directly to some GPIO pins and see if that works without generating the timeout crash.

Thank you...

joan2937, 2021-03-08:
I don't see how writing to GPIO 3 would cause any problem.

Please provide a complete but minimal script which we can run that demonstrates the problem.

rfresh737, 2021-03-09:
I found the problem. I had connected some wires to my breadboard hooking up another encoder. I must have knocked a wire loose going to the i2c pins. Just by luck, I happen to re-seat those i2c wires and things started to work normally again.

Sorry for the false alarm.

closed 2021-03-09 by guymcswain