joan2937/pigpio #438: How do I attach my encoder A pin to an interrupt from a MCP23017

rfresh737, 2021-03-09:
I'm trying to attach an encoder A pin to GPA5 pin on my MCP23017, and then setup the interrupt.

Below is my snippet of code. However, when I turn the knob, the interrupt does not fire.

I believe the problem might be because the mcp_pin5 assignment (commented below with *******) is not an integer type?

I'm using the Adafruit MCP230xx library.

Thank you...

`
def crs_encoder(gpio, level, tick):
	print ("crs_encoder fires")

i2c = busio.I2C(board.SCL, board.SDA)
mcp = MCP23017(i2c, address=0x20)

mcp_pin5 = mcp.get_pin(5) # ***********************
mcp_pin5.direction = digitalio.Direction.INPUT
mcp_pin5.pull = digitalio.Pull.UP

pi_crs = pigpio.pi()
if not pi_crs.connected:
	exit()

pi_crs.set_mode(mcp_pin5, pigpio.INPUT)
pi_crs.set_pull_up_down(mcp_pin5, pigpio.PUD_UP)
pi_crs.write(mcp_pin5, 1) # 1=high
cb1 = pi_crs.callback(mcp_pin5, pigpio.EITHER_EDGE, crs_encoder)
`

joan2937, 2021-03-10:
This is not a pigpio issue.

This seems to be a general programming question which is not the point of this site.

closed 2021-03-10 by joan2937

rfresh737, 2021-03-10:
Right, I agree. I was just looking to see if there were any examples
out there.