joan2937/pigpio #439: Can't connect to pigpio at localhost(8888) Can't create callback thread. Perhaps too many simultaneous pigpio connections

Madhuraank, 2021-03-11:
Code:
```
import subprocess
import time
#subprocess.call("sudo pigpiod", shell=True)
#time.sleep(0.1)
from pigpio_dht import DHT11, DHT22
while True:
        gpio = 4 # BCM Numbering

        sensor = DHT11(gpio)
#sensor = DHT22(gpio)

        result = sensor.read()
        print(result)

        temperature=([value for value in result.values()][0])
        print(temperature)

        humidity=([value for value in result.values()][2])
        print(humidity)
        time.sleep(10)

```

It works fine fine but after giving the result for few times then I get this error. New to pigpio please do help me figure out what's wrong
```
Output:

28
46

28
46

28
46

28
46

28
46
```
and then I get the following:
```

%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Can't connect to pigpio at localhost(8888)

Can't create callback thread.
Perhaps too many simultaneous pigpio connections.
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Traceback (most recent call last):
  File "dht8.py", line 9, in <module>
    sensor = DHT11(gpio)
  File "/home/pi/.local/lib/python3.7/site-packages/pigpio_dht/dht11.py", line 24, in __init__
    super(DHT11, self).__init__(gpio, pi=pi, timeout_secs=timeout_secs, use_internal_pullup=True, max_read_rate_secs=1, datum_byte_count=1)
  File "/home/pi/.local/lib/python3.7/site-packages/pigpio_dht/dhtxx.py", line 47, in __init__
    self._pi.set_pull_up_down(gpio, pigpio.PUD_UP)
  File "/usr/lib/python3/dist-packages/pigpio.py", line 1385, in set_pull_up_down
    return _u2i(_pigpio_command(self.sl, _PI_CMD_PUD, gpio, pud))
  File "/usr/lib/python3/dist-packages/pigpio.py", line 993, in _pigpio_command
    sl.s.send(struct.pack('IIII', cmd, p1, p2, 0))
AttributeError: 'NoneType' object has no attribute 'send'
```
Thanks in advance

guymcswain, 2021-03-11:
```python
while True:
        gpio = 4 # BCM Numbering

        sensor = DHT11(gpio)
```
My guess is this is creating infinite connections to pigpio.  Place this code outside the loop.

Madhuraank, 2021-03-12:
yeah  it worked..thanks

closed 2021-03-12 by Madhuraank