joan2937/pigpio #440: Raspberry Pi4 running Raspbian x64 - GPIOTEST

jhennessy-code, 2021-03-11:
When running GPIOTEST from within the terminal window. All sockets are reporting as failed:

pigpio -v = 79

Both running as sudo and standard user 

I am able to run commands such as ' gpio -g mode/write'

output:

Testing...
Write 1 to gpio 0 failed.
Pull up on gpio 0 failed.
Write 1 to gpio 1 failed.
Pull up on gpio 1 failed.
Write 1 to gpio 2 failed.
Pull up on gpio 2 failed.
Write 1 to gpio 3 failed.
Pull up on gpio 3 failed.
Write 1 to gpio 4 failed.
Pull up on gpio 4 failed.
Write 1 to gpio 5 failed.
Pull up on gpio 5 failed.
Write 1 to gpio 6 failed.
Pull up on gpio 6 failed.
Write 1 to gpio 7 failed.
Pull up on gpio 7 failed.
Write 1 to gpio 8 failed.
Pull up on gpio 8 failed.
Write 1 to gpio 9 failed.
Pull up on gpio 9 failed.
Write 1 to gpio 10 failed.
Pull up on gpio 10 failed.
Write 1 to gpio 11 failed.
Pull up on gpio 11 failed.
Write 1 to gpio 12 failed.
Pull up on gpio 12 failed.
Write 1 to gpio 13 failed.
Pull up on gpio 13 failed.
Write 1 to gpio 14 failed.
Pull up on gpio 14 failed.
Write 1 to gpio 15 failed.
Pull up on gpio 15 failed.
Write 1 to gpio 17 failed.
Pull up on gpio 17 failed.
Write 1 to gpio 18 failed.
Pull up on gpio 18 failed.
Write 1 to gpio 19 failed.
Pull up on gpio 19 failed.
Write 1 to gpio 20 failed.
Pull up on gpio 20 failed.
Write 1 to gpio 21 failed.
Pull up on gpio 21 failed.
Write 1 to gpio 22 failed.
Pull up on gpio 22 failed.
Write 1 to gpio 23 failed.
Pull up on gpio 23 failed.
Write 1 to gpio 24 failed.
Pull up on gpio 24 failed.
Write 1 to gpio 25 failed.
Pull up on gpio 25 failed.
Write 1 to gpio 26 failed.
Pull up on gpio 26 failed.
Write 1 to gpio 27 failed.
Pull up on gpio 27 failed.
Write 1 to gpio 28 failed.
Pull up on gpio 28 failed.
Write 1 to gpio 29 failed.
Pull up on gpio 29 failed.
Write 1 to gpio 30 failed.
Pull up on gpio 30 failed.
Write 1 to gpio 31 failed.
Pull up on gpio 31 failed.
Skipped non-user gpios: 16 
Tested user gpios: 0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 
Failed user gpios: 0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31

jhennessy-code, 2021-03-11:
Additional investigation has shown that my pigs hwver and pigs pigpv are both showing 0. Maybe this is a corrupt install?!

jhennessy-code, 2021-03-11:
This doesn't appear to be clearly documented any, total removal of get-apt version of pigio was required. In addition i had to remove the lib files. There is also additional file locations that were preventing the install of V79 which were listed upon running the make install command. I deleted these as i went until the command completed.
At this point we were back in action.

closed 2021-03-11 by jhennessy-code

guymcswain, 2021-03-11:
> removal of get-apt version of pigio was required

I think I've had this issue before.  It may be necessary depending on how your paths are configured, ie, look in /usr/local first I believe.  If you think it needs to be documented send me a proposed blurb to include.

Btw, what is GPIOTEST?