joan2937/pigpio #441: Suddenly, the infrared LED stopped working.

Light-dev-mac, 2021-03-13:
Excuse me.

I made an infrared remote control with Raspberry Pi 3 and it worked fine.
But suddenly it stopped working.
I bought a new hardware, but it doesn't heal, so
Probably due to the influence of the library. 

My Raspberry : Raspberry Pi 3 Model B Rev 1.2

# check C I/F

TEST  5.4  FAILED got 25 (callback: 50)
TEST  5.11 FAILED got -142 (wave tx busy, serial read: 0)

 # check C      I/F to daemon

TEST  5.4  FAILED got 26 (callback: 50)
TEST  5.10 FAILED got 1744 (wave tx busy, callback: 1702)
TEST  5.11 FAILED got -142 (wave tx busy, serial read: 0)

# check Python I/F to daemon

TEST  5.4  FAILED got 26 (callback: 50)
TEST  5.10 FAILED got 1733 (wave tx busy, callback: 1702)
bytearray(b'\x98\x80\xf8\x98\xfe\x9e~\x06\x98\x98\xe6\xf8\x06\x18~\x80\x9ef\x06\x98~\x86\x9e\xf8\x9e`fx\x18\x06\x98\x9ex\x06\x98\x9ef\x86\xf8\x06\x18x\x86\x9e\x1e\xe6\xe0\xe6\x9e\xf8\x9e`fx\xf8\x9e`\x86\x06\xe6\x98\x06\x06xf\x06\x98x\xe0\x9e\xfe\x9e\x18f\x9e\xfe\x9ef\xe6\xf8\x06\x98\xf8f~\xe6\x9e\xe6\x9ef\x86\xf8\x06\x18\xe0f\x06\x18~\x80\x9e\x86\x9e\x1e\x06\x98\xf8f~\xf8\x9e\x00\x98\x9ex\x06\x98x\xe6\x9e\x18\xe6\xe6\xe6~\x98\x80\x06\x86\x9e`\x06\x98\x80\x06\x9e\xe0\x9e\x00\x18~\x80\x9ef\x06\x98\xe0\x06\x9e\xfe\x9ef\x06x\x1e\x06\x18~\x80\x9e\x06\x06~\x00\x18\x9e\xfe\x9ef\x86\xf8\xe6\x18`\x06\x98\xe6\x06\xe6\x9e\xf8\x9e\x00\x98\x9ef\x86\xf8\x06\x18\x9e\xfe\x9ef\xe6\xf8fx\x98\x80\x86\x98\xf8\x9e\x00\x18~\x80\x9ef\x06\x18xffx\x00\x06\x18\xe0\xe6\x9e\x1e\xe6\x9e\xe6\x9e\x00\x98\x9ex\x06\x18~\x80\x9ef\x06\x98\x9e\x1efx\x06\x86\x9e\x00\x18\xe0f~\x18f\x9ef\x06x\xf8\x86\x98\x80\xf8\x98\xfe\x9e~\x06\x98\x80\x86\xf8fx\x00\x98\x9ef\x86\xf8\x06\x18\xe0\x86\xf8\xe6\x9e~\xe6\xf8\x06\x18\xe0\xe6\x9ef\x86\x9e`\x06\x98~\x86\x9e`\x06\x9e\x00\x18~\x86\x9e\x1e\x06~\xfe\x9e\x18f\x9e\xfe\x9ef\xe6\xf8\x06\x98~\x18fx\x06\x06~\x80\x9e\x1e\xe6~\x98\x80\xfe\x98f\x86\xf8\x06\x18\xe0\x86\xf8f~\x86\x9e\x1efx`\x06\x98\x80\x86\xf8f\x9e\x1e\x06\x18\x9ef\x86\x9e~\x06\x98\xe6\x06\x06\x18x\xfe\x9e\x18\x06\x98\x9e\xe6\x9e\xf8\x9eff\x9ef\x86\x9e`\xe6\xf8\xe6~\x98\x80\xfe\x98f\x86\xf8\x06\x98\xf8\x06~f\x86\xf8\x86\x9e\x00\x98\x80\x06\x9e\x06\x86\xf8f~\xe6\x9e\x1e\x06\x98\xe0\x06\x9e\x06\x86\x9e~fx`\x06\x18~\xfe\x9e\x00\x98\x9efx\x18\x86\xf8f\x06\x98\x9efxf\x06~\x86\x9e\xf8\x9e~\xe6\xf8\x06\x86\x98\x80\xfe\x98f\x86\xf8\x06\x18x\x18fx\x06\x06xxf~\xe0\x9e\x00\x98\x9ef\x86\xf8\xe6\xe0\x06\x9ef\xe6\xf8\x06\x18~\xfe\x9e\x00\x18xf\x06\x9e\x86\x9e~\x06\x9e`\x86xf\x06\x9e\x00\x98\x9efx\x06\xe6\xf8f~\x18fx\x1e\x86\x86\x98\x80~\x86\xf8f\x9e\xe6\x9e\xe6\x86xf\x9e\x1ef\xe6xf\x06x\x00\x98~\x06\x86\xf8\x06\x18\x9e\x06\x06~\x80\x9e\x00\x98\xf8f\x9e\xfe\x9e\xfe\x9e`\x06\x9e~\x18x\x00\x18\x9e\x86\x9e\x1e\x06\x98~\x18f\x9e\xf8\x9e\x9e\x9e\xe0\x9ef\x06x\x00\x18x\x18\xe6\x9e\xf8\x9e`\xe6~\x98\x80\x06\x86\x9e`\x06\x18\x9e\xfe\x9e~\x06\x86\x00\x98\x98\x86\x9e\x1e\x06~ff\x06x\x00\x98\x9ex\x06\x98\x9e\xe6\x9ef\x86\x9e`f\x9e\xf8\x9e~\x06\x18\xe0f\x86\xf8\x06xf\x06x\x00\x98\xf8\x06~ffx`\xe6\xf8\x86\x06`f\x9e\x00\x18x\x18f\x9e~\x06\x9e`\x06\x18~\x80\x9ef\x06\x98\xf8\xe6\x9ef\x06\x9e\x1e\x06\x98\x9ex\x06\x18xff\x86\xf8\x86xf\x06\x9e\x00\x98\x80\x06xxfx\x18\xe6\xf8f\x86\xf8f\x9ef\xe6\xf8\x06\x86\x98\x80\x80\x98f\x06\x98\xe0f\x06fx\x18\xe6\xf8\x06\x18\x9e\x86\x9e\xe6\x9e\x18\x06\x9e\x86\x06\x98\x98\x86\x9e\x00\x98\x80\x06\x18\x9e\x06\x06x\x86\xe6\x18\x1e\x06\x98\xe0\x06\x9e\x06f\x9e\x18fx\x18\x86\x06`f\x9e\x00\x18~\x80\x9ef\x06\x18\x9e\x06\xe6\xf8\xe6\xe0f\x9exf\x9e\xfe\x9ef\xe6\xf8\x06\x18\x06\x9eff\xe6\xf8f\x9e\xf8\x9e~\x06\x98\x9ex\x06\x98\x80\x06\x18\x9ef\x06~f\x86\x86\x98\x80') bytearray(b"\nNow is the winter of our discontent\nMade glorious summer by this sun of York;\nAnd all the clouds that lour\'d upon our house\nIn the deep bosom of the ocean buried.\nNow are our brows bound with victorious wreaths;\nOur bruised arms hung up for monuments;\nOur stern alarums changed to merry meetings,\nOur dreadful marches to delightful measures.\nGrim-visaged war hath smooth\'d his wrinkled front;\nAnd now, instead of mounting barded steeds\nTo fright the souls of fearful adversaries,\nHe capers nimbly in a lady\'s chamber\nTo the lascivious pleasing of a lute.\n")
TEST  5.11 FAILED got 0 (wave tx busy, serial read: 1)
TEST  5.26 FAILED got 26 (callback: 50)
TEST 13.7  FAILED got 2688 (glitch filter, wave chain, tally: 1792)
TEST 13.8  FAILED got 1792 (glitch filter, wave chain, tally: 896)
TEST 13.9  FAILED got 1792 (glitch filter, wave chain, tally: 0)
TEST 13.10 FAILED got 749 (noise filter, wave chain, tally: 1500)
TEST 13.12 FAILED got 13 (noise filter, wave chain, tally: 0)

# check pigs   I/F to daemon

Testing pigpio pigs
pigpio version 79
SLR-f fail with [23 230 158 134 6 24 158 6 102 158 102 6 152 152 230 248 6 24 230 230 158 6 134 158]
ERROR: No more CBs for waveform

# check pipe   I/F to daemon

SLR-f fail with (23 230 158 134 6 24 158 6 102 158 102 6 152 152 230 248 6 24 230 230 158 6 134 158)

Please help me.

guymcswain, 2021-03-16:
You may have a conflict in your system using the PCM peripheral.  You can change the default with a configuration option.  For C use [`gpioCfgClock()`,](http://abyz.me.uk/rpi/pigpio/cif.html#gpioCfgClock) for the pigpio daemon use the -t option.

Light-dev-mac, 2021-03-17:
Thank you.

I always  use pigpiod by  command  "sudo systemctl start pigpiod".
But, this command does not  allow "-t option "

So, I use  pigpiod by command "pigpiod -t 0"

But, my problem are not sloved.

I have another  question.
What's the difference between "systemctl start pigpiod" and  "pigpiod", terminal command.

Light-dev-mac, 2021-03-17:
Oh. Thank you . SOLVED!!!
"sudo pigpiod -t 0" make it well.
Thank you .

guymcswain, 2021-03-19:
Ok.  Thanks for letting me know.

closed 2021-03-19 by guymcswain