joan2937/pigpio #442: pigpiod  deamon start in root on boot up

PeteBear, 2021-03-15:
Hi, I am wanting to use pigpiod in node red. I have read all I can find however I still have issues of the deamon not starting on boot. I have modified rc.local and added the line as the instructionds say and when I run the following command the out put can be seen

pi@RefluxStill:~ $ sudo ps aux | grep pigpiod
pi 1074 0.0 0.0 7348 544 pts/0 S+ 20:25 0:00 grep --color=auto pigpiod

so the rc.local file has run it up in user pi however node red PWM output GPIO does not work, if i then type sudo pigiod and then run the above command again I get the following

pi@RefluxStill:~ $ sudo ps aux | grep pigpiod
root 1084 16.5 0.1 13568 1644 ? SLsl 20:34 0:00 pigpiod
pi 1091 0.0 0.0 7348 524 pts/0 S+ 20:34 0:00 grep --color=auto pigpiod

so if I understand I now have pigpiod running in root and in pi and node red works fine.
How can I get it to run in root at boot up as the software in node red will not work until it is running in root?

Thanks for your help in advance

guymcswain, 2021-03-15:
> pi@RefluxStill:~ $ sudo ps aux | grep pigpiod
pi 1074 0.0 0.0 7348 544 pts/0 S+ 20:25 0:00 grep --color=auto pigpiod

In the first instance, the process running is not `pigpiod` but `grep`.  So you don't have pigpiod running either as user pi or root.  Something must have gone wrong with your rc.local configuration.  Please show specifically changes you made to rc.local?

PeteBear, 2021-03-16:
Hi thanks for the email answered on line  - hope it helps you help me

Cheers

Pete

guymcswain, 2021-03-16:
> I appreciate your time in helping me could I hask you to explain what grep is and why it refferences pigpiod in the data returned?

`grep` attempts to search for a line of text that containing its argument.  In this case the lines of text are output from the `ps` command and the argument is "pigpiod".  `ps aux` will return status of all processes including the command line you just executed which happens to have the matching "pigpiod".   The output from that same command line would look as follows when pigpiod is running:
```bash
pi@rpi0ws40:~ $ sudo ps aux | grep pigpiod
root       523  9.0  0.4  11516  1908 ?        SLsl Mar15 201:13 pigpiod
pi        4473  0.0  0.4   6520  1840 pts/0    S+   13:53   0:00 grep --color=auto pigpiod
```

I don't use rc.local to run pigpiod as a service but use systemd instead, as follows:
### Running pigpiod as systemd service

- wget https://raw.githubusercontent.com/joan2937/pigpio/master/util/pigpiod.service
- sudo cp pigpiod.service /etc/systemd/system
- sudo systemctl daemon-reload
- sudo systemctl enable pigpiod.service
- sudo systemctl start pigpiod.service

closed 2021-03-19 by guymcswain