joan2937/pigpio #443: wave_create_and_pad returning unknown command

ExordiumAndTerminus, 2021-03-17:
I've run into a problem with pigpio.py, specifically the wave_create_and_pad method.
below is the code I've cobbled together for a test, which gives the resulting traceback below:

import pigpio
pi = pigpio.pi()
pi.wave_create_and_pad(10)
___
Traceback (most recent call last):
  File "thing.py", line 13, in <module>
    pi.wave_create_and_pad(10)
  File "/home/pi/.pyenv/versions/rPi/lib/python3.8/site-packages/pigpio.py", line 2352, in wave_create_and_pad
    return _u2i(_pigpio_command(self.sl, _PI_CMD_WVCAP, percent, 0))
  File "/home/pi/.pyenv/versions/rPi/lib/python3.8/site-packages/pigpio.py", line 1011, in _u2i
    raise error(error_text(v))
pigpio.error: 'unknown command'
____

Any idea on what is going wrong?  Going to those lines, I am seeing nothing amiss so I'm really confused as to what is broken here.

For further context: I have installed the latest version 1.78 from PyPi and was trying to run advpistepper, which utilizes this library, but was having issues and I tracked it down to those lines in particular.  By isolating the problem I was able to prove it is not an issue with advpistepper but with your library instead.

Thanks!
- Max

guymcswain, 2021-03-18:
Is this it?
```python
import pigpio
pi = pigpio.pi()
pi.wave_create_and_pad(10)
```
I never tested it without having a wave first initialized.  Does it fail after you add pulses?

ExordiumAndTerminus, 2021-03-18:
yes it does, the original code is taken from this and it appears they were following all of your steps only to still have this issue:
 def busy_loop(self):

        self.init_move()

        self.pi.wave_clear()
        prev_wave_id = -1

        # start of with a minimal delay pulse just so that we have a
        # current_wave_id. This saves one check in the loop.
        pulse = pigpio.pulse(0, 0, 100)
        self.pi.wave_add_generic([pulse])
        current_wave_id = self.pi.wave_create_and_pad(10)
        self.pi.wave_send_once(current_wave_id)
_______________
thanks for the speedy response by the way, it is much appreciated!

guymcswain, 2021-03-18:
My RPi4 runs this code without a problem.

```python
import pigpio
pi = pigpio.pi()

pi.wave_clear()
prev_wave_id = -1

# start of with a minimal delay pulse just so that we have a
# current_wave_id. This saves one check in the loop.
pulse = pigpio.pulse(0, 0, 100)
pi.wave_add_generic([pulse])
current_wave_id = pi.wave_create_and_pad(10)
pi.wave_send_once(current_wave_id)

print 'bye'
```

ExordiumAndTerminus, 2021-03-18:
are you using python 2 or 3?  I'm trying to run this with python 3 and that looks like python 2 syntax.  Could that be part of the issue on my end?

ExordiumAndTerminus, 2021-03-19:
Nevermind.  I got it.  Had to restart the pigpio daemon service after updating my pigpio package.  Thanks for the help!

closed 2021-03-19 by guymcswain