joan2937/pigpio #444: 64 bit Raspberry Pi OS

neil-benn, 2021-03-18:
Hello,

  I've got pigpio working well on the Raspberry Pi OS 32 bit; python3 libraries are found and no issues.  However, I will eventually need to use the 64 bit Raspberry Pi OS, which is currently in beta but working well so far.  I tried to pull down pigpio, make and make install it.  That all succeeded but when trying to run it I get the following:

pi@rpi-compute-1:~/pigpio-master $ sudo pigpiod
2021-03-18 18:26:11 initPeripherals: mmap gpio failed (Operation not permitted)
Can't initialise pigpio library

  The OS is defined as:

pi@rpi-compute-1:~/pigpio-master $ uname -a
Linux rpi-compute-1 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021 aarch64 GNU/Linux

  Would it need to do something to force the make to build against a 64 bit architecture?

  Thanks for your help.

Cheers,

Neil

guymcswain, 2021-03-18:
No it should work.  I was successful installing and running pigpio on PiOS64 a while ago.  Can you run 'pigpiod -v` to see which version of pigpio you have.

neil-benn, 2021-03-19:
It reports version 71

guymcswain, 2021-03-19:
You need to install a later version - latest is 79.  After installing, or prior to, you may need to `apt-get remove pigpio`.

neil-benn, 2021-03-22:
Thanks that works, previously I just pulled the head.

closed 2021-03-22 by neil-benn