joan2937/pigpio #445: Add SMI support to pigpio.

mfp20, 2021-03-23:
Hi,

In the past couple of days I've been trying to use the [Secondary Memory Interface](https://iosoft.blog/2020/07/16/raspberry-pi-smi/) to drive multiple stepper motors, reworking the [code given for neopixels](https://iosoft.blog/2020/09/29/raspberry-pi-multi-channel-ws2812/). I don't have a working example using DMA and PWM-paced SMI asserting DREQ... yet.

The SMI mode (ALT1) allows to use 8-9-16-18 pins in parallel and adjust the pulses with 4ns precision; with DMA, PWM and using the DREQ, preloading the data in memory, would allow to produce up to 18 pulse trains, perfectly in sync, perfect timing, zero cpu use.

I was searching for code creating a memory mapped device in linux filesystem and I've found servoblaster, so I remembered the credits on pigpio and after a second look at pigpio I realized that pigpio is already doing most of the job. So probably my time could be better spent in re-basing the code upon pigpio internals instead of some custom code, in order to help other users as well.

I've zero RPi experience and zero pigpio experience; I've just started to read pigpio code. So any tip would be appreciated.

guymcswain, 2021-03-23:
Greetings,

Yes with pigpio you can generate many DMA controlled PWM outputs.  The first consideration is the accuracy required by your application:  The timing on the pigpio PWM outputs is closer to 4 microseconds.  Also note that pigpio uses GPIO numbering based on the Broadcom specification which do not correspond to the actual pin locations on the GPIO connector on the RPi.
Good luck.

mfp20, 2021-03-23:
> Yes with pigpio you can generate many DMA controlled PWM outputs. The first consideration is the accuracy required by your application: The timing on the pigpio PWM outputs is closer to 4 microseconds.

I don't have a numeric accuracy in mind.

My target is to generate 250k pulses (2us high, 2us low) per second (or 500k time frames per second), on 16 pins; but I need to feed the fifo from some other peripheral (usb or wifi); and I'm afraid of having jitter because of the cpu is busy in receiving the data. A few uS of jitter each mS might be ok as well, as long as I can correct the error every now and then. Can I do 250 kHz jitter free/corrected, using current code or I need to enable the SMI?

In the case of SMI, the approach used in many projects is to insert a delay control block after each data control block, ie: after each pulse, the dma sends dummy data to PWM FIFO, the PWM delays the new data block to get the exact timing of the next pulse. The rawWaveXXX functions seems to be the proper hook to get a similar behaviour. Is it correct?

> Edit: I should also note that you can probably find a lot of help on the raspberrypi.org forum. pigpio is a collection of APIs that others have used to provide libraries that control stepper motors and such. Those are the folks who can probably best help you 

I wrote here to figure out whether adding SMI support to pigpio would be of interest or ... just one of my dumb ideas. I didn't get how to make a real time controller out of an rpi yet. Bare metal projects have limited peripheral support, and linux projects are mostly not-realtime; when the projects required strict timing (ex: radio/sound projects) they went for DMA+PWM/PCM. But those worked on a limited amount of pins; I need 16 pins.

Thanks for all the tips. I'll write a similar message on rpi forums to see If someone can add some valuable help!

guymcswain, 2021-03-23:
> My target is to generate 250k pulses (2us high, 2us low) per second (or 500k time frames per second)

The maximum update rate for DMA driving GPIO on pigpio is 1MHz so 500KHz is the maximum frequency one can generate.

> A few uS of jitter each mS might be ok as well, as long as I can correct the error every now and then. Can I do 250 kHz jitter free/corrected, using current code or I need to enable the SMI?

I don't know how to correct for jitter without adding external hardware and resampling but then you might as well use an external PWM generator chip.  

> I wrote here to figure out whether adding SMI support to pigpio would be of interest or ... just one of my dumb ideas.

Not dumb.  I've read the link you mentioned a while back and would love to experiment with SMI and pigpio but just never had the time.  

> The rawWaveXXX functions seems to be the proper hook to get a similar behaviour. Is it correct?

I don't know/have never used.  You will need to ask the developer @joan2937 .  If not here, Joan is usually responsive to questions about pigpio on the rpi forum.

mfp20, 2021-03-23:
> > My target is to generate 250k pulses (2us high, 2us low) per second (or 500k time frames per second)
> 
> The maximum update rate for DMA driving GPIO on pigpio is 1MHz so 500KHz is the maximum frequency one can generate.

Nice. What about the DMA uncached memory (mailbox) maximum size? Can I use large amounts (ex: 256MB) or there is some kind of addressing limit? I'm asking to know the maximum buffer size.

> I don't know how to correct for jitter without adding external hardware and resampling but then you might as well use an external PWM generator chip.

Indeed the RPis 0-4 have 2 more cores in the VC4, called VPUs, and 12 QPUs ... is it possible to use them somehow? In mailbox.h there are an execute_code() and execute_qpu(). But I couldn't figure out how to use those to run a callback or something similar.

> Not dumb. I've read the link you mentioned a while back and would love to experiment with SMI and pigpio but just never had the time.
>
>
> I don't know/have never used. You will need to ask the [...]

guymcswain, 2021-03-23:
Good questions for the forum - ie, I don't know.  Unless something has changed, the VC4 was never a public spec so your own your own there.

mfp20, 2021-03-24:
> Unless something has changed, the VC4 was never a public spec so your own your own there.

Everywhere is written that Broadcom released the VC4 docs in 2014, but I couldn't find it anymore.
GCC didn't include VC4 support, and now is stuck at gcc6 (or 7, can't remember); and nobody put together all the bits&bytes to make something like avr-libc for AVR. And all the reverse engineering efforts existing before the docs release ... disappeared.
The chip producer has gone open source but the community (!!!) didn't take advantage of that? Mystery.

> Good questions for the forum - ie, I don't know. 

Here I go a bit off-topic, about the forums in general. Feel free to ignore the rant.

To tell you the truth I'm a bit skeptical about those vendor forums. They tend to ... waste time. Let's say this way.
[Take as an example the one I wrote yesterday](https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=307736) cut&paste&gluing some of the sentences I wrote here a few hours before:

1) there's active moderation/censorship: my message had to be approved. ie: it isn't a safe environment. There might be the need of moderation as those forums are targeted by any kind of w-w-shit. But usually that's just an excuse for the vendor to drive the communities in its own direction, whether is business or an educational institution.

2) the first reply it's a kind of "life lesson" nobody asked for, without any kind of advice, help, tech detail. And ... it even doesn't look to be the real author, as browsing some of his previous 1587 messages... well... he never wrote so crappy messages. It looks like he smoked a ton of crack between his previous messages and the one in my thread. Or, it isn't him. And this takes us to the third point.

3) Those entities usually have (or allocate) little resources to manage the forums. This doesn't allow a proper moderation and even the basic security: avoid injections, mitm, identity theft and more in general manipulation of their users. Git-based tools like github offer superior security because they add git hashing on top of large, automatic, copy-on-write filesystems and related devops. Whatever off-side track manipulation can be detected easier.

4) Developers oriented envs are way more effective. Because we tend to focus on topic. A bit of kindness "hello", "regards", "cheers", a couple of jokes to raise some smiles, but most of the message is on topic. It makes our time better spent.

Maybe it is just my case, but my experience have been pretty negative in the past 10-15 years.

take care

closed 2021-03-24 by guymcswain