joan2937/pigpio #448: gpioCfgClock messing up /dev/serial0 comms

NBroomfield, 2021-04-03:
I added gpioCfgClock(5, 0, 0) to work around an issue with the library messing up the i2s peripherial which I'm using but now my comms via /dev/serial0 are messed up, any suggestions? I'm not actually using pigpio for anything complex just reading an writing to some pins, so I don't need complex PWM or interrupts on pins.

Thanks

guymcswain, 2021-04-03:
How are you using pigpio?  If using the daemon, restart the daemon using `sudo pigpiod -t 0`.

NBroomfield, 2021-04-03:
Thanks but I'm not using the daemon.

guymcswain, 2021-04-03:
Then did you set `gpioCfgClock()` accordingly?

NBroomfield, 2021-04-03:
Not sure? What would your recommend? Thanks :-)

guymcswain, 2021-04-03:
> I added gpioCfgClock(5, 0, 0) to work around an issue with the library messing up the i2s peripherial ...

Sorry, you have done so already.  I don't see how this has anything to do with the serial port.

guymcswain, 2021-04-03:
> I'm not actually using pigpio for anything complex just reading an writing to some pins, so I don't need complex PWM or interrupts on pins.

You may be interested in this library instead:
https://github.com/joan2937/lg

NBroomfield, 2021-04-03:
Thanks I'll take a look.

NBroomfield, 2021-04-06:
I think this was a red herring, please close.

closed 2021-04-06 by NBroomfield