joan2937/pigpio #450: error while loading shared libraries: ligpigpio.so on 64-bit RPi4

FilipSLO, 2021-04-03:
Hi joan2937.

First thanks for your great library. I used it for years from RPi3 onwards on Raspbian.
But on new install of [Mageia linux](https://www.mageia.org/) unfortunately it seems that  #284 doesn't help. At least I have this error on RPi4 4G:
`$ sudo ./x_pigpio
./x_pigpio: error while loading shared libraries: libpigpio.so.1: cannot open shared object file: No such file or directory
`
Same problem when I try to load pigpiod.

No recognizable warnings in build log.

Any idea?

closed 2021-08-16 by guymcswain

FilipSLO, 2021-08-18:
I would appreciate stated reason for closing.

guymcswain, 2021-08-18:
Yes, sorry.  We don’t have access to all the various os environments nor the time to test.  The os supported  for this library is the raspberry pi os.

FilipSLO, 2021-08-19:
Thanks for explanation. I might try again some day.

FilipSLO, 2023-04-15:
Thanks for posting your findings!

Maybe pigpio can be changed to search also /usr/local/.