joan2937/pigpio #451: TEST 13.9  FAILED got 1 (glitch filter, wave chain, tally: 0)

placheckij, 2021-04-08:
Hello,
First of all, great job on pigpio! Thank you very much for your work. Tested python. All other python tests passed. Used master branch, make & make install on raspberry pi zero wh headless via ssh (wifi connection). Have several things attached, nothing @GPIO25 tho:
GPIO18 - neopixels
GPIO19 - servo control
GPIO21 - input (no signal here during the tests).
Pi is powered via its +5V pin, both usbs and hdmi unoccupied, connected arducam motorized lens camera through the camera slot (also installed libarducam).
Regards,
Jakub

guymcswain, 2021-04-08:
Is the failure consistent?  Can you rule out the power supply by attaching usb power?

placheckij, 2021-04-12:
Just retried it on the same power source and it passed. No changes made to pi.

placheckij, 2021-04-15:
One time test failure, unable to reproduce.

closed 2021-04-15 by placheckij