joan2937/pigpio #452: Signal interference with Linux timer

berndporr, 2021-04-09:
I'm using the pigpiod API and that has no timer implementation. So I tried to use my own Linux timer: 
https://github.com/berndporr/cppTimer 
However, if I use that the program earlier or later hangs. It seems to be always when I use a pwm command. When I disable my Linux timer it's all good. So the implementation now is without the timer but a thread with a loop and a usleep which is not ideal.
Are all signals managed by pigpio(d) or are there some which are not managed by its signal handler?

berndporr, 2021-04-09:
I see that with the pigpio library I can disable signal handlers:
```
int cfg = gpioCfgGetInternals();
cfg |= PI_CFG_NOSIGHANDLER;  // (1<<10)
gpioCfgSetInternals(cfg);
int status = gpioInitialise();
```
Is that also possible for the daemon?

guymcswain, 2021-04-09:
I believe there is but don't see the option shown in the documentation.

guymcswain, 2021-04-09:
The -c option allows to set the internal options.  Set bit # 10.

berndporr, 2021-04-09:
Thanks! I'll do.

guymcswain, 2021-04-15:
@berndporr , is this issue still open?  Thanks for your other contributions.

berndporr, 2021-04-15:
oops. Indeed. Solved with the -c option. [ In the meantime I have switched to non-demon code but good to know for the future! ]

closed 2021-04-15 by berndporr