joan2937/pigpio #453: set_PWM_range returns real range and not just 0 or an error

berndporr, 2021-04-09:
set_PWM_range doc states:
```
Returns 0 if OK, otherwise PI_BAD_USER_GPIO, PI_BAD_DUTYRANGE,
or PI_NOT_PERMITTED.
```
https://github.com/joan2937/pigpio/blob/master/pigpiod_if2.h#L595
However the underlying function in pigpio.c returns the real range:
https://github.com/joan2937/pigpio/blob/master/pigpio.c#L9080
```
return pwmRealRange[gpioInfo[gpio].freqIdx];
```
The doc on the web page states the same. It obv makes sense to return the actual range which might be different to the one requested. I do a workaround for now in my code:
https://github.com/berndporr/alphabot/blob/main/alphabot.cpp#L9
where I just set the pwm range and then read it back. Either way the real range is really good to have as with that I can avoid rounding errors and get always the best resolution.

guymcswain, 2021-04-09:
I agree with you but if we change the API from the documentation then it could disrupt current users of the library.  It looks like you have a satisfactory working method so can this issue be closed?

berndporr, 2021-04-09:
The API is great as it is. I'd suggest to just fix the documentation and state that the function actually returns the real range not just zero on success. If you have no plans of changing it then I can actually use the return value. :)

berndporr, 2021-04-09:
BTW: it's correct for gpioSetPWMfrequency (non-demon API). Just the demon API description has this discrepancy. [ When I wrote this issue I was using the demon version but I've switched to the non-demon API now).

guymcswain, 2021-04-09:
Thank you for clarifying for me.  In that case it makes sense to change the doc to match the behavior.

commit de6064ce (2021-04-15): Fixed doc that set_PWM_range() returns the actual range

Fixed the documentation that the command set_PWM_range() actually
returns the actual range. This resolves #453.