joan2937/pigpio #454: gpioSerialRead reading hex data problem

radoslavilievm2m, 2021-04-09:
Hi,
I am trying to receive hex data over a bit-bang serial, but it seems to can transfer only text data with a \n delimiter. Also, I notified, that the count size is always one in this case, bc everytime it prints the sent pattern only one time, no matter how long is the sent string.
Do you have an idea how I can read the hex data.

Thank you in advance.
```
#include<stdio.h>
#include<stdint.h>
#include<string.h>
#include<unistd.h>
#include<pigpiod_if2.h>
//#include "pigpio.h"
#define UART1 21
#define UART2 6
#define UART3 7
#define UART4 8
 
 uint8_t buffer[256];
int main()
{
    int pi = pigpio_start(NULL,NULL);
    int i = 0;
    printf("%d",pi);
    set_mode(pi,UART1,PI_INPUT);
            printf("Hello world!");
bb_serial_read_open(pi,UART1,4800,8);
int count = 0;
    while(1)
    {
        count = bb_serial_read(pi,UART1,buffer,sizeof(buffer));
        for(i=0;i<count;i++)
        {
            printf("%s",buffer);
            memset(buffer,0,sizeof(buffer));
        }
        sleep(5);
    }
    return 0;
}
```

guymcswain, 2021-04-09:
I'm able to receive binary data but I'm using a different client.  Is your sender using standard 8N1?  It should work.

radoslavilievm2m, 2021-04-12:
I tried both with Tera Term and Termite, all of them using 8N1.
I think the printf maybe is the problem, because The "Hello world" on line 18 is printed after receiving the first string from UART.

guymcswain, 2021-04-15:
Print `count` the size of bytes received.  Also, you will not be able to see non-printable characters, ie, binary data values less than 30 or greater than 127, unless you using a different format specifier or write a function to convert each character to a "printable" hexadecimal value.

closed 2021-04-15 by guymcswain