joan2937/pigpio #458: error: 'F_SETPIPE_SZ' undeclared when compiling

nightrider77, 2021-04-25:
Hello.

Trying to compile pigpio for arm platform with uClibc toolchain.
Having error:

```
pigpio.c: In function 'gpioNotifyOpenWithSize':
pigpio.c:12029:21: error: 'F_SETPIPE_SZ' undeclared (first use in this function)
       i = fcntl(fd, F_SETPIPE_SZ, bufSize);
                     ^~~~~~~~~~~~
pigpio.c:12029:21: note: each undeclared identifier is reported only once for each function it appears in
Makefile:48: recipe for target 'pigpio.o' failed
```

Checked possible solution to add **#define _GNU_SOURCE**, but this file already has this line of code...

How to solve this?

Thanks.

nightrider77, 2021-04-25:
Temporarily fixed it with

`i = 1024 * 64;`

guymcswain, 2021-12-28:
Closing due to insufficient information to determine if an issue with pigpio exists.

closed 2021-12-28 by guymcswain