joan2937/pigpio #459: sigHandler: Unhandled signal 7, terminating

nightrider77, 2021-04-25:
Hello.

Arm system with uClibc (Linux 4.9.37 armv7l).

While starting daemon, I receive the following error:

```
# pigpiod
initAllocDMAMem: mbox open failed(No such file or directory)
Can't initialise pigpio library
# pigpiod -a1
# Unhandled fault: external abort on non-linefetch (0x008) at 0xb6fe2098
pgd = c5e44000
[b6fe2098] *pgd=85e56835, *pte=20101303, *ppte=20101a33
2021-04-25 23:24:03 sigHandler: Unhandled signal 7, terminating
```

guymcswain, 2021-04-26:
Which arm system?  What OS?

Pigpio is for Raspberry Pi and relies heavily on Raspberry Pi OS libraries to access built in peripherals and DMA capabilities.

nightrider77, 2021-04-26:
> Which arm system? What OS?
> 
> Pigpio is for Raspberry Pi and relies heavily on Raspberry Pi OS libraries to access built in peripherals and DMA capabilities.

LiteOS + Linux

guymcswain, 2021-04-28:
pigpio was designed specifically for the Raspberry Pi Arm based SOCs and is unlikely to work on anything else.

nightrider77, 2021-04-28:
Pity :(

closed 2021-04-28 by nightrider77