joan2937/pigpio #460: Feature Request:  Filter Notifications by Edge

jmbldwn, 2021-04-27:
The current notification protocol in `pigs` doesn't support selecting the edge of interest, so if you only care about falling or rising edges, you are getting twice the traffic over the socket connection.

This would be an optimization of course.  But it's not only the socket that would benefit, the daemon itself would deal with fewer interrupts if they are configured for only one edge.  Right now I see the daemon taking about 6-7% of a RPI3B+ CPU, which isn't a deal killer, but seems like more than necessary.

joan2937, 2021-04-27:
Could you clarify what you mean by notifications?  Some example code might be useful.

The daemon does not use interrupts.  Linux interrupts are too slow and are per GPIO.  If pigpio used interrupts it would be using 100% CPU time and capture far less edges.

pigpio uses DMA to sample/timestamp GPIO 0-31 every 5µs (by default).  The 6-7% CPU is going through those samples to detect level changes.  There would be no gain in filtering.

jmbldwn, 2021-04-27:
I see.  I was thinking for one or two GPIOs the interrupt method would be more efficient, but I take your word for it that with a larger number of active pins the interrupts become less efficient than periodic sampling.

I'm specifically thinking about the socket protocol between the daemon and its clients.

By notifications, I mean the messages sent to daemon clients when a transition is detected on a GPIO pin.  Currently, I get messages on both rising and falling edges, even if I only need one edge.  This means we're sending twice the traffic on the socket.  I am ignoring the notification I don't need on the receiving side, so it's wasted bandwidth.

I was thinking the NB command could be augmented to specify edge(s), which would tell the daemon to only transmit notifications over the socket on events we care about.
```
NB h bits <edges>
```
Again, this is an optimization, and the implementation may imply that we don't get much out of this.

guymcswain, 2021-04-28:
> I was thinking for one or two GPIOs the interrupt method would be more efficient, but I take your word for it that with a larger number of active pins the interrupts become less efficient than periodic sampling.

pigpio has APIs allowing you to use interrupts if you desire.  I think you originally were using the nodejs pigpio-client which did not implemented those APIs but it would be far easier to do that than to optimize pigpio's DMA based sampling engine.

jmbldwn, 2021-04-28:
I had assumed `pigpio-client` could only support features the daemon offers via the `pigs` socket api.

Are you saying the daemon always sends all samples over the socket?

guymcswain, 2021-04-28:
No, your right, only SIF API are supported.

jmbldwn, 2021-04-28:
OK, so this suggestion was to augment the SIF API to support filtering by edge, then `pigpio-client` would have to be upgraded to support it.

If it were me, I'd table it until there's some other reason to open up the SIF API.