joan2937/pigpio #461: pigpiod unable to initialize with kernel 5.11

GlenPickle, 2021-05-05:
Kernel 5.11 introduced some big changes to how GPIOs are accessed. When moving to kernel 5.11 on Ubuntu 21.04 (armhf) I get the following errors when attempting to start pigpiod:

```
ubuntu@ubuntu:~$ sudo pigpiod -v
79
ubuntu@ubuntu:~$ sudo pigpiod
ubuntu@ubuntu:~$ 2021-05-05 16:54:19 initAllocDMAMem: mbox open failed(No such device or address)
Can't initialise pigpio library

ubuntu@ubuntu:~$ sudo pigpiod -a1
ubuntu@ubuntu:~$ 2021-05-05 16:54:26 sigHandler: Unhandled signal 11, terminating
```

Since this works on Ubuntu 20.10 with kernel 5.8 I believe the kernel changes are causing this issue.

guymcswain, 2021-05-05:
Pigpio does not use the kernel for GPIO access.

GlenPickle, 2021-05-05:
What else should I look into to determine why I'm having these failures on the latest Ubuntu?

joan2937, 2021-05-05:
Well the failure is as given in the error message

```2021-05-05 16:54:19 initAllocDMAMem: mbox open failed(No such device or address)```

Presumably the mailbox has been renamed or removed in Ubuntu.

joan2937, 2021-05-06:
@GlenPickle Could you add the command to your comment for others who come across the issue?

GlenPickle, 2021-05-06:
Absolutely, the command is `sudo modprobe vcio`. This will eventually be fixed in Ubuntu 21.04 by automatically loading that module at boot time.

joan2937, 2021-05-06:
@GlenPickle Thanks.