joan2937/pigpio #462: PIGPIO and UART

caseypen, 2021-05-05:
Hi,
I'm doing a project using Rpi-0w. The Rpi reads two Hx711 with serial communication. I modified the code from [the example](http://abyz.me.uk/rpi/pigpio/examples.html#Python%20code) provided by pigpio. The Rpi went well if I only read the two load cells for over 5 hours. At the same time, I use UART from Rpi to read the messages from another device. The Rpi also went well if I only read the UART messages for over 5 hours. 
However, if I have the two modules (UART reading and HX711 reading) work at the same time, the system works fine at the first hour, but the UART gets much noise data after that, which makes the UART communication not be able to get the data in an ideal frequency. However, I cannot find any errors from the system. Can you provide any suggestions for debugging this issue? I really have no idea where the error comes from.

Thanks and regards,

Chen

guymcswain, 2021-07-05:
@caseypen , Sorry for the late response but this description didn't provide anything specific enough to act upon.  If you believe there is an issue with pigpio library you will need to provide some code to demonstrate the issue.

guymcswain, 2021-08-16:
Closing due to inactivity.

closed 2021-08-16 by guymcswain