joan2937/pigpio #466: What this fix implemented?  In version 78 or 79?  I just encountered it on Pi OS buster Debian 10 (fully patched as of today).  Happen to be running 78.

dachshund-digital, 2021-05-13:
What this fix implemented?  In version 78 or 79?  I just encountered it on Pi OS buster Debian 10 (fully patched as of today).  Happen to be running 78.

May 03 00:12:29 pi3modelb0 pigpiod[32482]: 2021-05-03 00:12:29 initInitialise: bind to port 8888 failed (Cannot assign requested address)
May 03 00:12:29 pi3modelb0 pigpiod[32482]: Can't initialise pigpio library
M

_Originally posted by @dachshund-digital in https://github.com/joan2937/pigpio/issues/298#issuecomment-830949171_

joan2937, 2021-05-14:
Initialisation has failed because some other process is already using port 8888.  You need to find out which process is using port 8888.

dachshund-digital, 2021-05-15:
That is the not case, I confirmed via netstat that such was not happening.

guymcswain, 2021-05-15:
@dachshund-digital please change the title of this issue as it is confusing - refers to a closed issue which was fixed as far back as version 74.

I see that Debian 10.9 was released end of March.  I will try to upgrade one of my platforms to see if I can reproduce.  Any additional information would be helpful.

dachshund-digital, 2021-05-16:
Just what I showed... that sometimes the daemon stop is not clean, i.e. the port seems to be put into limbo, can't be required as expected is what it seems like.  I have seen this most often when I do a compile from source, upgrading from say 78 to 79 for example.  Before I run the new version, I typically do a systemctl stop pigpiod (which stops 78), which should release all resources, right?  Then when I do a systemctl restart pigpiod (which starts 79) I see the above issue.  Humm.... does make install stop and start any existing variant of the daemon?

guymcswain, 2021-05-16:
I don't believe make touches anything dealing with systemd.  The package manager may have a stale `pigpiod.service` file.  Check it against the one in the repo.

dachshund-digital, 2021-05-16:
Yeah, and we worked through a lot of issues with the unit file some time ago.  We added an improved PID kill command to make sure at system shutdown and reboot, the daemon did not hold up the sequence.  That has been working well actually thus far.

closed 2021-08-16 by guymcswain