joan2937/pigpio #468: pigpiod provokes system reboot

WimGo, 2021-05-19:
When running gpiotest, the RPi4 reboots.
[image]

I installed pigpiod in two ways:

1. 
```
wget https://github.com/joan2937/pigpio/archive/master.zip
unzip master.zip
cd pigpio-master
make
sudo make install
```

2. `sudo apt-get install pigpiod`

This lead to two versions of the program. But they present the same issue.
```
$ ls -latr /usr/bin/pigpiod  /usr/local/bin/pigpiod
-rwxr-xr-x 1 root root 9784 Mar 30 16:42 /usr/bin/pigpiod
-rwxr-xr-x 1 root root 9808 May 17 16:44 /usr/local/bin/pigpiod
$ sudo /usr/bin/pigpiod -v
79
$ sudo /usr/local/bin/pigpiod -v
79
```
This is my HW and SW config:
```
$ cat /etc/debian_version
10.6

$ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

$ uname -a
Linux rpi4 5.4.72-v7l+ #1356 SMP Thu Oct 22 13:57:51 BST 2020 armv7l GNU/Linux

$ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1 -> same data
processor       : 2 -> same data
processor       : 3 -> same data

Hardware        : BCM2711
Revision        : d03114
Serial          : 10000000ba0ac550
Model           : Raspberry Pi 4 Model B Rev 1.4
```

joan2937, 2021-05-19:
Don't run gpiotest.

What do `pigs pigpv` and `pigs hwver` report?

WimGo, 2021-05-19:
```
$ pigs pigpv
79
$ pigs hwver
13644052
```

joan2937, 2021-05-20:
I see nothing there that should cause any problem.

Are you sure that nothing is connected to the GPIO?

WimGo, 2021-05-20:
Quite sure: I saw that nothing should be connected to the GPIO pins. So I had removed the connector all together.
[image]

joan2937, 2021-05-20:
I suggest you manually do part of what gpiotest does and see where it fails.

for ((g=2;g<28;g++)); do echo "test gpio $g"; pigs w $g 0 r $g w $g 1 r $g m $g r pud $g u r $g pud $g d r $g; done

WimGo, 2021-05-20:
This seems to have worked correctly. At least no crash or reboot:
```test gpio 2
0
1
1
1
test gpio 3
0
1
1
1
test gpio 4
0
1
0
0
test gpio 5
0
1
1
0
test gpio 6
0
1
1
0
test gpio 7
0
1
1
0
test gpio 8
0
1
1
0
test gpio 9
0
1
1
0
test gpio 10
0
1
1
0
test gpio 11
0
1
1
0
test gpio 12
0
1
1
0
test gpio 13
0
1
1
0
test gpio 14
0
1
1
0
test gpio 15
0
1
1
0
test gpio 16
0
1
1
0
test gpio 17
0
1
1
0
test gpio 18
0
1
1
0
test gpio 19
0
1
1
0
test gpio 20
0
1
1
0
test gpio 21
0
1
1
0
test gpio 22
0
1
0
0
test gpio 23
0
1
1
1
test gpio 24
0
1
1
0
test gpio 25
0
1
1
0
test gpio 26
0
1
1
0
test gpio 27
0
1
1
0
```

WimGo, 2021-05-20:
I continued testing using piscope, in combination with this script.
Also did I reconnect the gpio flat cable and tested the 3V3 - gpio24 connection via a pushbutton. piscope also shows this correctly.
So the conclusion is that I should not have used the gpiotest shell script to find out is pigpiod runs well. It does run well using other interfaces, as used by piscope and pigs.
Thanks for your support.

closed 2021-05-20 by WimGo

joan2937, 2021-05-20:
I guess the Pi is being misidentified and during gpiotest an invalid GPIO is being changed.

It is probably incorrectly testing one of GPIO 0-1, or 28-31.

If you want to check (using GPIO 28 as an example) see if each of those GPIO report the following.

```
$ pigs w 28 1
-41
ERROR: no permission to update GPIO
```

joan2937, 2021-05-20:
Actually what GPIO does piscope show?  My Pi4B shows 2-27.

WimGo, 2021-05-21:
[image]

WimGo, 2021-05-21:
With pigs, all seems correct. Cf. above and below.
```
$ pigs w 28 1
-41
ERROR: no permission to update GPIO
$ pigs w 1 1
-41
ERROR: no permission to update GPIO
$ pigs w 0 1
-41
ERROR: no permission to update GPIO
```

joan2937, 2021-05-21:
Oh well.  It's a bit of a mystery then.