joan2937/pigpio #469: mock mode for testing client apps

xloem, 2021-05-20:
I'm just adding this because it would help encourage stable software, and it doesn't seem like it would be hard to add.

It would be great if pigpiod supported some kind of mock server mode, where it didn't actually write or read any gpios and didn't even need to be running on a raspberry pi.  This would make it much easier for people writing gpio apps to write tests for their code.

guymcswain, 2021-05-21:
> This would make it much easier for people writing gpio apps to write tests for their code.

What is making it difficult when developing with the actual firmware/hardware?   

> it doesn't seem like it would be hard to add.

That depends on the scope.

In general, I think mocks are useful as a standalone piece of code.  I don't quite understand how a mode can be added to pigpio that would allow it run on a different machine.  pigpio is very tightly coupled to the RPi platform.

xloem, 2021-05-21:
It's somewhat reasonable with real hardware, but for cross-platform apps a raspberry pi may not always be available.

I just implemented a prototype gnuradio block to connect to pigpiod; to let the block be tested across upgrades and platforms to make sure nothing broke I actually wrote a small python pigpiod simulator.  But with my code being different from yours the test will have its edgecases wrong etc.

I looked into mutating your code a little: one approach was having file-backed memory instead of accessing /dev/mem.  Another approach was adding checks to the command handlers to see if simulation mode was active.

It may be that a far better solution to this problem is to emulate a raspberry pi with qemu.

closed 2021-08-16 by guymcswain

xloem, 2021-08-16:
Thanks for keeping your project maintained and organised.  I'm sad this was closed as projects where nobody has a pi could add support for pigpio with a way to test.

I looked into using qemu to emulate a pi and it is pretty nonintuitive.  The situation was kind of a single poc with no general setup instruction.

Would you guys put my python script into some kind of contrib-like folder if I dug it up?

guymcswain, 2021-08-16:
If you have something that you think would be useful to others AND you are willing to support it, then yes.