joan2937/pigpio #470: Sensor readout while wave_tx_busy()

philippjunger, 2021-05-21:
Dear all,

first of all I want to thank you joan for creating this amazing PIGPIO lib!
I am using it to create a nice ramp for my stepper motor, using a TMC2208 (Trinamics) motor driver.

After some troubleshooting I managed to get the PIGPIO work as I want it to.
Now I want to read out my "homing" sensor while the stepper is moving to check if a limit was reached.

I am using the following code to read out the sensor while the chain is in progress.
The function "checkLimits will return true if a sensor is active.

```
# Check if limits are reached during movement
        while (pi.wave_tx_busy()): # While transmitting.    
            if (checkLimits(axis,steps,PUL)):
                pi.wave_tx_stop()
                break
            sleep(0.1)
```
The problem is, that my PWM is looking really bad when this while loop is active and the motor is doing noisy movement.
Even if I just enter:

```
# Check if limits are reached during movement
        while (pi.wave_tx_busy()): # While transmitting.    
		sleep(0.1)
```

It behaves the same. It feels like the while loop is disturbing the wave chain progress.
Is there any better way to read out the sensor after hitting the pi.wave_chain(chain) command?

Thank you very much, I really appreciate your help!
Philipp

guymcswain, 2021-05-21:
When you just play the waveform, without any looping, is your stepper motor performing noise/jitter free?

Heavy network traffic on the pi will induce some additional jitter - on the order of 3-4 microseconds - on top of a baseline of 2-3 us jitter.  (These were my observations I made quite some time ago.)

philippjunger, 2021-05-24:
When just playing the wave, the stepper motor is doing fine without any problems.

guymcswain, 2021-05-25:
> The problem is, that my PWM is looking really bad when this while loop is active...

Is there any way you can quantify the problem and provide a script that we can use to reproduce?

philippjunger, 2021-05-25:
So this is a direct comparison of noice of the two movements. The first is with the "while" part and the second is without it.
The motor is also not at the correct position at the end.

https://user-images.githubusercontent.com/74316765/119476267-31785300-bd4e-11eb-89e8-880b580643b5.mov
https://user-images.githubusercontent.com/74316765/119476239-2a514500-bd4e-11eb-89ae-f225904cfae0.mov

guymcswain, 2021-05-25:
Since I don't have a stepper motor handy, I will be looking at the output pin with an oscilloscope to see any distortion that might be the cause of the motor noise.  If you have such a scope trace it would nice to see it for comparison.

Also, why the `pi.set_PWM_dutycycle(PUL, 128)` before transmitting the wave chain?  PWM will be turned off whenever waves are transmitting.

guymcswain, 2021-08-16:
Closing due to lack of response.

closed 2021-08-16 by guymcswain