joan2937/pigpio #471: Debugging Wave Output Without Scope

xloem, 2021-05-21:
I'm trying to send out a square wave using pigs and don't appear to be getting the right output.  Is there a way within pigpio to see what is happening, without having to connect an oscilloscope of some kind?

This is working for me to generate a square wave, which is audible as a noise as I hear my device on a relay rapidly toggling:
```
for ((i=0; i<100; i++)); do pigs w 4 1; pigs w 4 0; done
```

This is not working for me; I do not hear the device toggling, which indicates a long delay is present that I don't expect:
```
pigs wvag 4 0 25000 0 4 25000
pigs wvcap 50
pigs wvtxr 0
```

Additionally, this is not working for me; I do not see any output while either of the other commands are running:
```
pigs no
pigs nb 0 0xffff
pigs2vcd /dev/pigpio0
```

joan2937, 2021-05-21:
For a visual use piscope.

http://abyz.me.uk/rpi/pigpio/piscope.html

For a listing look at monitor.py

http://abyz.me.uk/rpi/pigpio/examples.html#Python_monitor_py

When using waves make sure you set any GPIO you use to outputs.

N.B. waves use bits to identify the GPIO (remember multiple GPIO may be set/cleared in each wave triplet).

pigs m 4 w

pigs wvag 16 0 25000 0 16 25000

Generally for GPIO x.

G=((1<<x))

pigs wvag $G 0 25000 0 $G 25000

xloem, 2021-05-21:
Thanks so much, it was totally using bits in the wave generation.

closed 2021-05-21 by xloem

xloem, 2021-05-21:
@joan2937 so you know, monitor.py also does not work on output pins.  it seems it is best to do some wiring if debugging on output pins is needed.

joan2937, 2021-05-22:
@xloem monitor.py works on all GPIO regardless of their mode.  Why do you think it does not work on output GPIO?

xloem, 2021-05-22:
I received no output for pins set to output, when I drove them, same as the behavior of the notification events above.

joan2937, 2021-05-22:
@xloem If you have time can you give an example showing the problem.  I'll try it on my Pi.