joan2937/pigpio #473: Using shell() always fails (Python API)

arc12, 2021-05-24:
Even simple cases return with code 512 and have no effect, such that I suspect the script is not actually executing.

Script created in /opt/pigpio/cgi and "chmod +x" applied.

By "simple cases", I mean things like ls, echo appending to a tmp file. Tried with and without a bash shebang, with and without parameters.

I'm running from a remote machine and connecting to the daemon, which works beautifully except for shell() being obstinate! Using Raspbian GNU/Linux 10 (buster).

An issue or user error?

joan2937, 2021-05-24:
It works for me with pigs.

I created `q`

```bash
#!/bin/bash
echo "hello world" >/tmp/pigpio.cgi
```

```bash
sudo chmod +x /opt/pigpio/cgi/q`
pigs shell q eee`
cat /tmp/pigpio.cgi
hello world
```

arc12, 2021-05-25:
Thanks for replying. I cant see how what I did differed from your example, but clearly something was awry.

closed 2021-05-25 by arc12

joan2937, 2021-05-25:
If you still have a problem we would need the shell script and the Python script you use.  Also details of how you are running the Python (e.g. Python version, remote or local etc.)