joan2937/pigpio #474: pigpio_if Error Messages - PIGIF_UNCONNECTED_PI and PIGIF_TOO_MANY_PIS Lack Message

dhrynyk, 2021-06-04:
Hi:

In pigpio_if.h two error codes are defined as
   pigif_notify_failed      = -2009,
   pigif_callback_not_found = -2010,

However, in pigpio_f.c in the function 'char *pigpio_error(int errnum)' these two error codes have been omitted. Are these errors no longer relevant?

I am basing the question on the downloaded release 79.

Thank you,
David

joan2937, 2021-06-04:
Hi David.

That looks like a mistake on our part.  I can think of no reason for their omission as they can still be raised.

joan2937, 2021-06-04:
Actually I spoke too soon.  pigpiod_if.h/c is deprecated and is not being kept up to date.

You should be using pigpiod_if2.h/c.

dhrynyk, 2021-06-04:
Thank you. I should have paid closer attention to the examples where the
includes state pigpiod_if2.h.

I appreciate the clarification,
David

closed 2021-06-05 by joan2937