joan2937/pigpio #475: Unexpected WAIT behavior in scripts

DanH42, 2021-06-19:
I'm just getting started with writing scripts for pigpio, and I ran into some unusual behavior with the `wait` instruction. In a nutshell:

- After initializing an input pin, `wait` does not wait for the pin to change state
- Calling `wait` twice in a row behaves the way that I would expect a single `wait` to work

For simplicity, the following examples were all run while no other scripts were running, and with nothing connected to pin 17. Pin 17 is the only pin I'm waiting on (`1 << 17 == 131072`). I'm seeing the same behavior on other pins whether or not they're connected to things. I'm running the latest v79 release.

If I initialize a pin as an input, pull it high, then wait for changes, like so:

    ~# pigs proc m 17 R pud 17 U wait 131072
    0
    ~# pigs procr 0; sleep 1; pigs procp 0
    1 0 0 0 0 0 0 0 0 0 0

...the script continues immediately, which in this minimal example causes the script to finish.

In fact, it proceeds so quickly that the results are the same even if I completely remove the `sleep` and do everything in a single `pigs` command:

    ~# pigs procr 0 procp 0
    1 0 0 0 0 0 0 0 0 0 0

...the script has already finished executing by the time I'm able to check its status.

Through some trial and error, I found that I could work around this behavior by simply waiting twice in a row:

    ~# pigs proc m 17 R pud 17 U wait 131072 wait 131072
    1
    ~# pigs procr 1; sleep 1; pigs procp 1
    3 0 0 0 0 0 0 0 0 0 0
    ~# pigs procs 1

...the script status is now `waiting`. This is the behavior I want, but it's not the code that I'd expect would achieve this behavior.

Armed with the knowledge that a second wait works, I tried simply waiting a bit after initializing the pins. However, even several seconds after doing so, the next `wait` still doesn't wait:

    ~# pigs proc wait 131072
    2
    ~#  pigs m 17 R pud 17 U; sleep 10; pigs procr 2; sleep 1; pigs procp 2
    1 0 0 0 0 0 0 0 0 0 0

What's going on here? Is this a bug, or am I just missing something about how `wait` is supposed to work? If it's the latter, would it be safe to rely on this double-wait behavior being consistent, or is there something else I should be doing? Since I don't understand the underlying cause, I'm concerned that blindly waiting twice in a row could cause me to miss edges if the first `wait` ever did actually wait.

guymcswain, 2021-06-30:
> Is this a bug, or am I just missing something about how wait is supposed to work? 

Likely a bug as I interpret the documentation the same as you have.

As a maintainer but not the author, I'm reluctant to modify this code because of the thread mutex logic involved.  There are other ways to obtain pin events without using pigs and scripts.  If you want to eliminate network latency the best way is to write a c program linking pigpio.  If you stay with a scripting language such as python and run on localhost you eliminate any external network dependencies.  Maybe if you describe what you want to accomplish we can provide you with some alternatives.

closed 2021-08-16 by guymcswain

DanH42, 2021-08-22:
> Maybe if you describe what you want to accomplish we can provide you with some alternatives.

I can try!

First, for some context: I've got a bunch of Pi Zero W's scattered around the house doing various things. Some Pis are single-purpose, others do several unrelated things on the same board. Some things only happen on one Pi, others run on multiple hosts at once. To keep things sane, each Pi has a single instance of pigpiod, and each application runs in its own Docker container and talks to the daemon for any of its GPIO needs. This setup has worked well for me, so long as I don't need super precise timing, which I mostly don't.

However: I've got some 4x4 matrix keypads that I need to scan for key presses. That involves setting one axis as an input, then once I see a key press, quickly switching the other axis to inputs so I can read a row/column pair. That operation needs to finish before the key is released, which makes fast timing important. I initially wrote all this logic in Javascript, my preferred language, and got it working from there. Even though the network I'm talking on is all on the local device, it can still take ~100 milliseconds to receive a pin change event, read the state of those pins, reconfigure everything, and then read the pins again. It's still pretty quick and works as long I'm deliberate about firm key presses, but it means that if I get a bit lazy and only give a button a quick tap, it may not register because the button just wasn't held down quite long enough.

A pigpio script seemed like the perfect solution to this problem, so I came up with this:

```
TAG 0 # start

# Set X axis to input pullup
M   ${pinNumbers.X1} R
M   ${pinNumbers.X2} R
M   ${pinNumbers.X3} R
M   ${pinNumbers.X4} R
PUD ${pinNumbers.X1} U
PUD ${pinNumbers.X2} U
PUD ${pinNumbers.X3} U
PUD ${pinNumbers.X4} U

# Write lows to the Y axis
M ${pinNumbers.Y1} W
M ${pinNumbers.Y2} W
M ${pinNumbers.Y3} W
M ${pinNumbers.Y4} W
W ${pinNumbers.Y1} 0
W ${pinNumbers.Y2} 0
W ${pinNumbers.Y3} 0
W ${pinNumbers.Y4} 0

# Wait for a keypress
WAIT ${gpioBitmask}
# Wait again, for some reason?
WAIT ${gpioBitmask}

# Read bank 1, store in p0
BR1
STA p0

# Write lows to the X axis
M ${pinNumbers.X1} W
M ${pinNumbers.X2} W
M ${pinNumbers.X3} W
M ${pinNumbers.X4} W
W ${pinNumbers.X1} 0
W ${pinNumbers.X2} 0
W ${pinNumbers.X3} 0
W ${pinNumbers.X4} 0

# Set Y axis to input pullup
M   ${pinNumbers.Y1} R
M   ${pinNumbers.Y2} R
M   ${pinNumbers.Y3} R
M   ${pinNumbers.Y4} R
PUD ${pinNumbers.Y1} U
PUD ${pinNumbers.Y2} U
PUD ${pinNumbers.Y3} U
PUD ${pinNumbers.Y4} U

# Read bank 1, store in p1
BR1
STA p1

# Pull signal pin high for 1us
W ${signalPinNum} 1
MICS 1
W ${signalPinNum} 0

JMP 0 # goto start
```

There's some Javascript string interpolation in there, and I'm using a small preprocessor that lets me add comments. The output that actually gets sent to the daemon looks like this:

```
TAG 0 M 19 R M 13 R M 6 R M 5 R PUD 19 U PUD 13 U PUD 6 U PUD 5 U M 21 W M 20 W M 16 W M 26 W W 21 0 W 20 0 W 16 0 W 26 0 WAIT 532576 WAIT 532576 BR1 STA p0 M 19 W M 13 W M 6 W M 5 W W 19 0 W 13 0 W 6 0 W 5 0 M 21 R M 20 R M 16 R M 26 R PUD 21 U PUD 20 U PUD 16 U PUD 26 U BR1 STA p1 W 12 1 MICS 1 W 12 0 JMP 0
```

Back over in Javascript land, I use [pigpio-client](https://www.npmjs.com/package/pigpio-client) to store and start that script, which isn't supported natively by the library, but I patched in some rudimentary support for it locally (and also did some extra testing to be reasonably sure that it's not my patched-in support that's causing trouble; I do see this same behavior when I do things manually through the `pigs` CLI). That script configures itself to wait for a keypress, and then when it sees one, it quickly grabs two copies of GPIO bank 1, one for each axis, which it stores in `p0` and `p1` to make them externally accessible. Then it toggles pin 12, an empty pin that I'm just using for internal signaling (there may be a cleaner way of doing this), to notify my application that there's something for it to see. My application reads out the two bank snapshots, which it can compare to tell what button(s) is being pressed.

(Hopefully this was enough to give a clear idea of what I'm doing; I can put together a fully functioning repo with a quick wiring diagram if you'd prefer.)

The good news is that it does exactly what I want most of the time, and it's _substantially_ faster. I'm now unable to physically press and release a button faster than it can be read, which is good enough for me. The bad news is... it doesn't always work, and sometimes I end up missing key presses. I haven't been able to characterize what's somehow different in those situations where it doesn't work, so all I can really say is that double-waiting isn't a silver bullet, but I don't know why. It doesn't seem to correspond to how quickly or how long I press a button.

So in a nutshell: All of this work is happening on the same device, but in containers that still talk via TCP, so the latency can add up. I need a performance boost for a specific task, but I'd like to keep the all-Docker architecture I've standardized on, and scripts seem like the best way to do that. I can switch to Python if there's a reason to, but I generally prefer to avoid writing in C. I have no idea what I'm doing in regards to pigpio, so if I'm missing something else that would be simpler, I'm open to alternatives.

guymcswain, 2021-08-22:
Can we reduce this down to the following problem statement:  “How to scan a 4x4 matrix keypad using pigpio daemon?

If so, I can share my thoughts but I would suggest the raspberrypi.org forums is a good place to pose this question.  You’ll get a broad spectrum of viewpoints that could be educational.  

Btw, the matrix keypads that I’m aware of don’t require pins to be switched from input to output.  So if you have a non genetic device please share the specific requirements.

DanH42, 2021-08-23:
> Can we reduce this down to the following problem statement: “How to scan a 4x4 matrix keypad using pigpio daemon?

I don't think so. I'm already able to scan these keypads just fine without involving pigpio scripts at all. It may be possible to improve the performance of my original approach a bit more, but the main purpose of this issue was to figure out if the specific behavior I was seeing in the `WAIT` instruction is a bug or if I was just misunderstanding how it's supposed to work. It sounds like it's a bug, so that would seem to deserve an open issue. The keypad thing happens to be the first use case I've run into for scripts, but there may be others in the future, and I can't imagine many GPIO-bound things I'd be doing that _wouldn't_ involve waiting for a pin to change state.

> Btw, the matrix keypads that I’m aware of don’t require pins to be switched from input to output (or vice versa). So if you have a non genetic device please share the specific requirements.

There's nothing special about the keypads I have, but the only no-reconfiguration approaches I've seen for scanning a matrix keypad (including many on the raspberrypi.org forums) have involved repeatedly polling rows one at a time for button presses, which sounds worse for both resource usage and responsiveness -- I'm trying to handle a relatively rare edge case where a key is pressed and released in under ~100ms, so scanning the keypad 10x/s would be a tie for responsiveness while forcing the Pi to work a lot harder. If you're aware of an event-driven approach that doesn't involve reconfiguring pins after initial setup, that would be great: not only would it probably simplify my code a bit, but I'll bet it would also work around this issue entirely, since `WAIT` only seems to behave unusually the first time it's called after a pin's been reconfigured.

That said, I'm also fine to just leave my code as-is, since the script-free version works fine the majority of the time, and keep discussion here to the core `WAIT` issue itself.

guymcswain, 2021-08-23:
> but the only no-reconfiguration approaches I've seen for scanning a matrix keypad (including many on the raspberrypi.org forums) have involved repeatedly polling rows one at a time for button presses, which sounds worse for both resource usage and responsiveness > 

I’m unsure what is meant by reconfiguration but pigpio can scan row (output) pins without consuming cpu cycles then send your application notifications of what is occurring on the column (input) pins without the use of scripts and the WAIT instruction.

DanH42, 2021-08-24:
What I meant by "polling rows" is that the solutions I've seen that don't involve switching pin modes have to read one row/column _at a time_. Here's one such (non-pigpio) example that I think is roughly what you're describing:

```
def readLine(line, characters):
    GPIO.output(line, GPIO.HIGH)
    if(GPIO.input(C1) == 1):
        print(characters[0])
    if(GPIO.input(C2) == 1):
        print(characters[1])
    if(GPIO.input(C3) == 1):
        print(characters[2])
    if(GPIO.input(C4) == 1):
        print(characters[3])
    GPIO.output(line, GPIO.LOW)

while True:
    readLine(L1, ["1","2","3","A"])
    readLine(L2, ["4","5","6","B"])
    readLine(L3, ["7","8","9","C"])
    readLine(L4, ["*","0","#","D"])
    time.sleep(0.1)
```

This method doesn't require changing pin modes, but it does require polling each row one at a time, which means you can't take advantage of pigpio's event notifications. If you pulled all the rows high at once, you wouldn't be able to tell which row was pressed, only which column.

I suppose I could pull all 4 rows high, wait for a pin notification, quickly scan all 4 rows individually, then pull them all high again and continue waiting, but that scan still involves tight timing. Probably even tighter timing actually, since I now have to take 4 separate readings instead of just 2, which probably means I'd still want to use a script to ensure that I can scan all 4 rows before a button is released. That would, however, probably work around this WAIT bug, so it may be worth a shot anyway at some point to see if the pros outweigh the cons. A secondary objective of this project was to just familiarize myself with pigpio scripts for future projects, and this would be a way of doing that.

guymcswain, 2021-08-24:
> but it does require polling each row one at a time, which means you can't take advantage of pigpio's event notifications. 

This is not so.  Unless you deliberately disable the alert monitoring thread, pigpio will continuously monitor the gpio pins.

guymcswain, 2021-08-24:
So here is my recommended approach to scan a keypad using pigpio:

- Use waveforms to generate the patterns required to 'poll' the row pins.  
- Set up notification callbacks for each row and col pin of the keypad.
- Maintain a state variable that indicates which row is currently active.  This var is updated by the callback associated with each row pin.  It is also accessible by the column pin callback such that when a column becomes active you know which row is driving this event, ie, you know the key that is getting pressed.

The above is all done using pigpio's DMA engines so only requires initial one time set up.  The logic in your app's callbacks sort out which key  is pressed.  Simple and elegant.  In this case, scripts are a distraction from utilizing the powerful features of the pigpio library.