joan2937/pigpio #476: [Question] : UART supported by this library ?

Noschvie, 2021-07-02:
Hello
Does this library support the UART interface of the "PN532 NFC HAT" ?
Thanks
Norbert

guymcswain, 2021-07-04:
From what I gather from a quick search on this part - PN532 NFC HAT - you just need to connect with it using a UART.  You don't really need pigpio to do that unless you wish to attach it to a virtual UART (using the bit-bang serial APIs) on any available GPIOs.  However, a virtual UART is not high speed and may not be appropriate for your application.

Note, that the pigpio serial APIs actually use the linux device driver at /dev/tty* or /dev/serial* and do not access the hardware UART peripheral directly.

Noschvie, 2021-07-06:
Maybe the I2C interface is prefered ?

closed 2021-08-16 by guymcswain