joan2937/pigpio #477: Aperiodicity/distortion of sipmle periodical signal

tanshihaj, 2021-07-04:
I've noticed that on my Raspberry PI Zero W signal generated with commands

```
pi@raspberrypi:~ $ sudo pigpiod
pi@raspberrypi:~ $ G=23; pigs m $G w; BG=$((1<<G)); pigs wvag $BG 0 10 0 $BG 10; pigs wvcre; pigs wvtxr 0
```

Looks like that on oscilloscope:
[image]

or like that (distortion changes from time and sometimes disappear):
[image]

I've tried to change power cable and power adapter but issue remains. How can I find out where is the problem?

guymcswain, 2021-07-05:
A few observations:

1)  Your script assumes the `wid` of the waveform to transmit is always 0.  This is only the case after you first start up the daemon or after each `pigs wvclr`.

2) You are running a waveform repeatedly and it needs to be stopped explicitly before another wave can run.

guymcswain, 2021-08-16:
Closing issue due to lack of response.

closed 2021-08-16 by guymcswain