joan2937/pigpio #478: Qt RPI4  gpioTerminate() Hangs when called

cogthought, 2021-07-06:
Hi,

When gpioTerminate() is called on a RPI4 using Qt5.15 on app exit, the apps hangs and does not quit.  However, 'ctrl - z' or a kill -9 of the process can quit the app.  

Is this expected behavior?  

When should  gpioTerminate() be called - if called at all?

Does pigpio work on RPI4.

app is linked against the pigpio library.

Thanks for the help,
Robert

guymcswain, 2021-07-07:
> When gpioTerminate() is called on a RPI4 using Qt5.15 on app exit, the apps hangs and does not quit. However, 'ctrl - z' or a kill -9 of the process can quit the app.
> 
> Is this expected behavior?

This function resets the used DMA channels, releases memory, and terminates any running threads.  It has no return value.
> 
> When should gpioTerminate() be called - if called at all?

After the library is initialized (`gpioInitialise()`) but before your application exits.
> 
> Does pigpio work on RPI4.

Yes

**Please note:**
_If you intend to rely on signals sent to your application, you should turn off the internal signal handling as shown in this example:_
```c
int cfg = gpioCfgGetInternals();
cfg |= PI_CFG_NOSIGHANDLER;  // (1<<10)
gpioCfgSetInternals(cfg);
int status = gpioInitialise();
```

cogthought, 2021-07-07:
Hi Guy,

Thanks for the quick response.

I am using signals in my app, and I did turn off signal handling in
piqpio.  I can gpioInitialise() okay; however, whenever I call
gpioTerminate() the apps hangs and will not respond to anything but a
"ctrl-z" (becomes a zombie) or "kill -9" which exits the app. "ctrl-c" has
no effect.

Thanks,
Robert

cogthought, 2021-07-07:
Hi Guy,

Another follow up note.  As long as I never call gpioInitialise();, my app
exits as expected.

Regards,
Robert

cogthought, 2021-07-07:
Hi Guy,

Version is reported as 79.

Regards,
Robert

guymcswain, 2021-07-07:
> whenever I call gpioTerminate() the apps hangs and will not respond

How are you calling?

cogthought, 2021-07-07:
Hi Guy,

I call it at the end of main.  See below.

int main(int argc, char *argv[]) {
       qDebug() << "[main] Running in GUI mode";
       operationalMode = GUI;
       MainWindow w;
       w.show();

       int retValue = a.exec();
#if 1
       // call on program exit
       gpioTerminate();
#endif
       return retValue;

I found that anytime I call gpioInitialise() the app will hang and not exit
no matter if gpioTerminate() is called or not.

I found that if I use pigpiod and link against pigpiod_if2, the app exits
without issue.  The problem only happens when I link against pigpio.

Thanks,
Robert

guymcswain, 2021-07-07:
When you link against pigpio do you have root privilege.

cogthought, 2021-07-07:
Hi Guy

I'm cross-compiling on a Ubuntu machine using QtCreator and deploying to
the RPi4.  I execute with root privilege locally the RPi4.

Does QtCtreator need root privileges during compile?

I have verified that even though the app hangs at exit, it can still read
and write both spi interfaces on the RPi4 and I can see the data on my
logic analyzer, so as long as the app does not exit, it works as expected.

There is an option in the app for the user to exit back to the OS if
needed..

Thanks,
Robert

guymcswain, 2021-07-07:
> I execute with root privilege locally the RPi4.

That's what I was looking for.   Thanks.

I don't understand Qt so the code snippet above doesn't tell me when gpioTerminate gets called - is it a signal?   If it's a SIGTERM maybe pigpio is still consuming it and you code doesn't see it.

cogthought, 2021-07-07:
Hi Guy,

Thanks again for your help.  it is called at the last point in Qt where it
exits.

While the apps is running, if I sent it a 'ctrl-c' my signal handler picks
it up.  See below:
^C
Caught Signal(2)
Removing PID File
Shutting down ...

But the app never shutdowns.

guymcswain, 2021-07-08:
Is that the pigpio pid file you’re removing?

felixkasza, 2021-07-08:
Out of curiosity, what exactly does that handler do? When I hear "signal handler", I mostly think "Thou Shalt Now …!" with a long liost of things that are forbidden and a _very_ short list of things that a signal handler may do.

cogthought, 2021-07-08:
Hi Guy,

No. It is the app's pid.  The app creates a pid file on launch/start.
Other applications can find the pid file, read the pid from the file and
then send signals to the app's pid.

Handy for doing app updates, etc.

Thanks,
Robert

cogthought, 2021-07-08:
Hi Felix,

Controls the apps.  Stops the app, and can be used for verify the app is
running.  The pid file only exists while the app is running.

If you look in /var/run on the RPi4, you will find pid files of other apps.

Regards,
Robert

felixkasza, 2021-07-08:
> Controls the apps. Stops the app, and can be used for verify the app is running.

Thanks. I should have worded my question differently:

You are aware, I presume, that signal handlers must be fully reentrant and can therefore only call functions and APIs that are themselves guaranteed to be reentrant. That applies even if the handler ultimately terminates the process.

Generally, I stick with Michael Kerrisk's advice: block signals, then have what amounts to a mini-server running on its own thread in a corner of the process that uses signalfd() and epoll(). If it is necessary to let the main part of the application know about something, either stuff it into a sig_atomic_t (or std::atomic_t<>) and find a place where you can hook into QT's main event loop.

felixkasza, 2021-07-08:
But going back to the beginning, you posted a fairly simple main(). If I fire up your application and then do the equivalent of "File > Exit", the a.exec() (*w*.exec()?) call should return to main, gpioTerminate() gets called, no signals involved at all, yet gpioTerminate still hangs.

Is that correct?

And if yes, are there any pigpio alerts, callbacks, notifications or whatnot still active (have not been explicitly cancelled)?

guymcswain, 2021-07-08:
I’m interested in a response to @felixkasza ‘s last question.

Also, turn on user level debug after gpioInitialize.  Use gpioCfgGetInternals, gpioCfgSetInternals to set bit 2.

cogthought, 2021-07-08:
Hi Guy and Felix,

I modified the mainwindow to just run gpioCfgSetInternals(cfg) with debug
enabled; gpioInitialise(), and run a simple spi test. When tested, the
system failed to exit.

Good News -> I then disable signal handling in the app and pid file
creation, and the system exited as expected.

I then re-enabled pid file creation, enabled the full application, removed
gpioCfgSetInternals calls, and the system exited under user control as
expected.

So the issue appears to be the way pigpio handles signals.  For this
application, I don't need signal support because I'm using zmq for messages
to the app externally.  Plus with the pid file, I can still kill the app if
needed externally.

Thanks again for the help.

Thanks,
Robert

guymcswain, 2021-07-09:
Hi Robert,

I'm glad you got your app working.  For me to continue working this as an issue for pigpio I have some follow up questions and will need more information such as debug logs or a simplified code snippet that demonstrates an issue.

> I modified the mainwindow to just run gpioCfgSetInternals(cfg) with debug
enabled; gpioInitialise(), and run a simple spi test. When tested, the
system failed to exit.

**Here pigpio has its signal handler disabled.   Does this mean your app didn't call `gpioTerminate`?   The debug output would be useful.**

> Good News -> I then disable signal handling in the app and pid file
creation, and the system exited as expected.

**So if neither pigpio nor your app is handling the signal how does the app exit?**

> I then re-enabled pid file creation, enabled the full application, removed
gpioCfgSetInternals calls, and the system exited under user control as
expected.

**In this case pigpio is handling signals and will call `gpioTerminate` on any signal other than SIGUSR1, SIGUSR2, SIGCHLD, SIGWINCH.  A debug trace would reveal if pigpio signal handler caught the exception.**

Thanks, 
Guy

cogthought, 2021-07-09:
Hi Guy,

I'll run the tests you requested and get the results to you.  I can also
send you a project to compile and test, but you will need Qt 5.15 libraries
set up on the  RPi.  If needed, I could forward you various Qt library
files needed for the project to set up on the RPi.

I have not tried it yet, but this failure may be reproduced on other
platforms.

Thanks,
Robert

guymcswain, 2021-07-09:
> I'll run the tests you requested and get the results to you

Let's just start with this.  I don't think I'll have the time/patience to learn and compile a Qt project :)

cogthought, 2021-07-09:
Hi Guy,

Will do.  I understand.

Thanks,
Robert

cogthought, 2021-07-14:
Hi Guy,

Sorry for the late reply with the data requested. Below are
example application runs with and without app signal handlers enabled.

Command used to turn on debug [is this correct]:
int cfg = gpioCfgGetInternals();
cfg |= PI_CFG_NOSIGHANDLER | (1<<2);  // (1<<10)
gpioCfgSetInternals(cfg);

The Qt App call this function to monitor signals:
catchUnixSignals({SIGQUIT, SIGINT, SIGTERM, SIGHUP});

Example App output with signals enabled [hangs]:
sudo ./temptrax.sh
[07/14/21, 14:35:58.449] Debug: [main] Running in GUI mode
[07/14/21, 14:35:58.497] Debug: [MainWindow::MainWindow] gpio test begin
*** INPUT DMA CONTROL BLOCKS ***
i=0 s=0 d=0 len=0 s=0 nxt=0
i=0 s=0 d=0 len=0 s=0 nxt=0
... [^^^ shortened, over 400K lines printed with the same message]
[07/14/21, 14:37:57.775] Debug: "[MainWindow::MainWindow] gpioVersion
pigpio = 79"
[07/14/21, 14:37:57.776] Debug: "[MainWindow::MainWindow] SPI Handle = 0"
[07/14/21, 14:37:57.777] Debug: 0           0 <- [simple SPI example, not
device connected]
[07/14/21, 14:37:57.780] Debug: 0           0
  <--- app does not exit
^C
Caught Signal(2)  <-- app signal handler enabled
Removing PID File
Shutting down ...   <--- Hang
Killed  <-- Kill -9

Example App output no signals enabled [exit normally]:
[07/14/21, 14:35:58.497] Debug: [MainWindow::MainWindow] gpio test begin
*** INPUT DMA CONTROL BLOCKS ***
i=0 s=0 d=0 len=0 s=0 nxt=0
i=0 s=0 d=0 len=0 s=0 nxt=0
... [^^^ shortened, over 400K lines printed with the same message]
[07/14/21, 14:53:46.877] Debug: "[MainWindow::MainWindow] gpioVersion
pigpio = 79"
[07/14/21, 14:53:46.878] Debug: "[MainWindow::MainWindow] SPI Handle = 0"
[07/14/21, 14:53:46.878] Debug: 0           0   <- [simple SPI example, not
device connected]
[07/14/21, 14:53:46.879] Debug: 0           0
<-- exit okay [no signal handlers installed]

Please let me know if there are any questions or if you need any
other information.

Thanks again for all the help.

Thanks,
Robert

guymcswain, 2021-08-16:
> Command used to turn on debug [is this correct]:

> int cfg = gpioCfgGetInternals();
 cfg |= PI_CFG_NOSIGHANDLER | (1<<2);  // (1<<10)
 gpioCfgSetInternals(cfg);

Yes but why did you not set bit 10 when your app is using signals?  I think it should have been set on the first example and off for the second example.

guymcswain, 2021-12-28:
Closing due to insufficient information to determine if an issue exists with pigpio.

closed 2021-12-28 by guymcswain

wschopohl, 2023-01-06:
Hi, I seem to have a similar issue using pigpio on a raspberry 4 with buildroot aarch64 custom image.
When trying to pinpoint the error, I started with the pigpiod application. My understanding is, that when I start the pigpiod and send it a SIGHUP or SIGTERM signal, it should just exit normally. But it doesn't. pigpiod daemon also seems to freeze at this point. It seems like there is some thread started by pigpio that's just reluctant to let go..

guymcswain, 2023-01-10:
I made a significant change to the library's signal handling a while back but just now merged it into the `develop` branch.  Can you retest with this branch to see if you observe the same behaviour?

wschopohl, 2023-01-14:
I cloned the dev branch and cross compiled pigpiod.c but the behavior hasn't changed. Thanks for your input though.

wschopohl, 2023-01-16:
After investigating a bit more with the develop branch and my limited gdb skills, it seems like the fifo interface is causing problems. When I start `pigpiod` with the `-f` option, I can stop it with a SIGTERM signal.

That means in my program I can use this code to disable the fifo interface and everything works as expected:

`gpioCfgInterfaces(PI_DISABLE_FIFO_IF); // call before gpioInitialise()`

guymcswain, 2023-01-16:
@wschopohl 
Thank you for the feedback on the develop branch.  Can you please open a new issue that describes your specific usage of the library?