joan2937/pigpio #479: Simultaneous waveforms on different pins?

faceless2, 2021-07-19:
I'm using pigpio C interface to communicate with multiple DALI lighting buses, and I'm hitting issues when I use more than one. Timing here is +/- 40us for each pulse, so pretty sure waves are my only option.

Each bus uses a pair of GPIO pins for RX and TX. For each bus, when I want to send I create a waveform:
```c
void send(int val) {
    gpioPulse_t pulse[bits];
    memset(pulse, 0, sizeof(pulse)); 
    // populate pulse here
    gpioWaveAddGeneric(bits, pulse);
    int waveid = gpioWaveCreate();
    gpioWaveTxSend(waveid, PI_WAVE_MODE_ONE_SHOT);
}
```
I'm calling this from within the callback to `gpioSetAlertFuncEx`.

My mental model of this was each time I call `gpioWaveTxSend`, the new wave is interleaved with any waves that are currently transmitting on other pins.  So while I'm managing the timings for each bus to ensure I don't transmit one wave until the previous has finished, I'm ignoring any waveforms transmitting on other pins.

This does't seem to be how it works, because if I call the above function to send on pin 2 while a wave is already transmitting on pin 1, it goes wrong (and the fact `gpioWaveTxAt` returns a single waveid implies that only one can transmit at a time).

So my questions are:

1. How does a second call to `gpioWaveTxSend` interact with any wave currently transmitting?
2. What is the correct approach for interleaving independent waves on different pins? Specifically, if a wave is transmitting on pin 1 and I want to start a new wave transmitting on pin 2, how do I do this without waiting for the first wave to finish?

If the answer to 2 is "interrupt the current waveform and recreate it with what remains of the first waveform, then add the second waveform", then more questions:

3. Assuming there is no waveform currently transmitting, if I call `gpioWaveTxStart` can I reasonably assume it's going to start transmitting immediately(ish)? Or if not, is there some way of determining the tick when the waveform starts transmitting other than listening for the pin change?
4. Same question as above, but what if I'd just called `gpioWaveTxStop` to cancel the current waveform? Is that going to introduce any sort of delay if I then immediately call `gpioWaveTxStart`?

guymcswain, 2021-07-19:
> How does a second call to gpioWaveTxSend interact with any wave currently transmitting?

The second wave cannot transmit until the first wave has completed or is explicitly stopped.  You can synchronize the two waves such that the first ends on the same ‘tick’ as the second begins.

> What is the correct approach for interleaving independent waves on different pins?

I think you want to synchronize the waves.  Break your waves into equal time segments and transmit them synchronously.  The padded wave apis are useful for transmitting a continuous wave.

faceless2, 2021-07-19:
Thanks for the impressively quick response Guy - I've been investigating doing this asynchronously but hadn't considered a rotating buffer of synchronized waves - so when I receive a new transmit request, I can just push it into the next one. 

I'll give it a try. Appreciate your help!

closed 2021-07-19 by faceless2

faceless2, 2021-07-19:
@guymcswain just wanted to say thanks - thats now, finally, working. For anyone else browsing the issue list, as I've been, here's my posterity post:

I ended up calling something like this from the GPIO alert callback - done there because I have a 1ms watchdog timer going there anyway, and the protocol means I can only buffer about 3ms of traffic. This is essentially the example from the `gpioWaveCreatePad` function
```c
int wid = -1, nextwid = -1
gpioPulse_t pulse[8];

void nextBlock() {
    if (wid < 0 && nextwid == wid) {
        // initialize
        for (int i=0;i<LEN;i++) {
            pulse[i].usDelay = 416;
        }
        gpioWaveAddGeneric(LEN, pulse);
        wid = gpioWaveCreatePad(50, 50, 0);
        gpioWaveTxSend(wid, PI_WAVE_MODE_ONE_SHOT);
        nextwid = wid;
    }
    if (nextwid == wid) {
        // here is where I shift in the next block of LEN bits into pulse
        gpioWaveAddGeneric(LEN, pulse);
        nextwid = gpioWaveCreatePad(50, 50, 0);
        gpioWaveTxSend(nextwid, PI_WAVE_MODE_ONE_SHOT_SYNC);
    }
    int currentwid = gpioWaveTxAt();
    if (currentwid == PI_NO_TX_WAVE) {
        fprintf(stderr, "Too slow!\n");
        gpioWaveDelete(wid);
        wid = nextwid;
    } else if (currentwid == PI_WAVE_NOT_FOUND) {
        fprintf(stderr, "ERROR: wavetx got WAVE_NOT_FOUND\n");
    } else if (currentwid == nextwid) {
        gpioWaveDelete(wid);
        wid = nextwid;
    }
}
```
My main issue was latency - I was constantly seeng "Too slow", which means the current wave had ended before the next one started. This on a RPi3B+ running this task on the standard kernel, and not much else.

The solution was to mess with the scheduler adding this to main:
```c
struct sched_param param;
param.sched_priority = 1;
if (sched_setscheduler(getpid(), SCHED_FIFO, &param)) {
    perror("sched_setscheduler");
}
```

Although the four transmit pins do not need to be synchronized, doing so and writing them out in one batch made the writing a lot simpler, so that was good advice.

Thanks again!