joan2937/pigpio #480: [Feature] Add support for operating systems that can not properly provide gpioHardwareRevision

raballew, 2021-07-22:
I am running a PXE booted custom alpha build of [Flatcar Container Linux](https://kinvolk.io/flatcar-container-linux/) on my Raspberry Pi 4 which is basically a Gentoo live operating system. This means, I am not installing anything and the Raspberry get netbooted every reboot. In this particular setup the gpioHardwareRevision can not be properly determined:

In `/proc/cpuinfo` none of the keys `Hardware`,  `Revision`, `Serial` or `Model` are present.

```text
core@localhost:~# cat /proc/cpuinfo 

processor       : 0
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
BogoMIPS        : 108.00
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3
```

From my understanding due to the nature of the netboot, the file `/proc/device-tree/system/linux,revision` is not present on the system. This results in [rev](https://github.com/joan2937/pigpio/blob/c33738a320a3e28824af7807edafda440952c05d/pigpio.c#L13785) always being 0 and `pigpiod` not starting properly.

**Usecase**

My idea is to run a Kubernetes Cluster of Raspberry Pi's while still being able to use the GPIOs but from within the cluster. This approach would allow me to run a `sensor` cluster while using all the benefits Kubernetes offers as a resource scheduler and manager.

**Solution**

I would propose as a solution to this issue, that users can specify an environment variable to manually set the value for those usecases. This would work both, for the systemd unit running the daemon or of `pigpiod` was started from command line.

guymcswain, 2021-07-22:
Why not put the missing information in the file /proc/cpuinfo?

raballew, 2021-07-22:
@guymcswain If I remember correctly, `/proc/` filesystem is a virtual one, the files in there aren't real and the filesystem as a whole is simply an interface to some of the information the kernel is holding in memory. You can't change `/proc/cpuinfo` unless you want to modify the kernel. If you could, your changes would be wiped out every time the file was updated by the kernel. So basically there two options, either I modify the kernel so that it prints the right information or `pigpio` offers a way to override the path where to get the information from.

Anyway, I was raising this request to open a discussion if that even possible or out of scope for `pigpio`.

guymcswain, 2021-07-22:
https://www.raspberrypi.org/forums/viewtopic.php?t=232417&start=25

this thread contains a discussion on how to bind mount a fake /proc/cpuinfo

guymcswain, 2021-08-16:
A solution was offered that does not require a change to the library.

closed 2021-08-16 by guymcswain