joan2937/pigpio #485: Interrupt not working as interrupt

0U55AMA, 2021-08-15:
Hi every one,
 I am running a JavaScript code on raspberry pi , in my code there is interrupt function from (pigpio)  , i tested detection and it works , but it is not working as interrupt , it does not interrupt any thing , if the code running a function and the interrupt function call nothing will happen until the function finish !!!  

`const Gpio = require('pigpio').Gpio;
const rpio = require('rpio') // just for delay 

var j = 0;

const button = new Gpio(15, { mode: Gpio.INPUT,pullUpDown: Gpio.PUD_DOWN,edge: Gpio.FALLING_EDGE});

for (let i = 0; i < 32; i++) {
console.log("a " + i);
rpio.sleep(5)    }

button.on('interrupt', (level) => {
  console.log(j+=1)
  rpio.sleep(5)    });

// while(1){}`

guymcswain, 2021-08-16:
After I familiarized myself with the rpio library the issue becomes obvious.  The rpio.sleep function is blocking.  Besides that, you should not mix two gpio libraries.  Either use rpio exclusively, which appears to support interrupts, or use the internal javascript function - setTimeout.

0U55AMA, 2021-08-16:
I removed the rpio.sleep function , and try again , the same problem  **it does not interrupt** (see CODE and Result) , [rpio module for JavaScript](https://www.npmjs.com/package/rpio) does not support interrupt , i cant use setTimeout because i need hardware interrupt .

**CODE** : 
`
const Gpio = require('pigpio').Gpio;
var i =0 ;

const button = new Gpio(15, {mode: Gpio.INPUT, pullUpDown: Gpio.PUD_DOWN, edge: Gpio.FALLING_EDGE});

for (let i = 0; i < 50000; i++) {
console.log("a" + i);}

button.on('interrupt', (level) => {
  console.log(i+=1)});
`

**Result** :
...
a49995
a49996
a49997
a49998
a49999
1
2
3
4
5
...

**Note** : interrupt is called so early before  the ("a" + i) for loop ends , but nothing is happen until this loop ends ,,  then (i+=1) works ... **this is not interruption at all !**

guymcswain, 2021-08-16:
Instead of complaining you should try to understand why.  JavaScript is single threaded.  Cardinal rule 1) Don’t block the event loop.

closed 2021-08-16 by guymcswain

felixkasza, 2021-08-16:
> for (let i = 0; i < 32; i++) {
> console.log("a " + i);
> rpio.sleep(5) }
> 
> button.on('interrupt', (level) => {
> console.log(j+=1)
> rpio.sleep(5) });

I am not big on Javascript, but — aren't you finishing the "a1" to "a49999" loop before you even _enable_ the interrupt?

That said, most languages and execution environments do not support interrupts, and even asynchronous callbacks are mostly unsupported. Pigpio itself operates in polling mode.

You may want to consider a background thread that either queries pigpio from time to time (ew) or that asks pigpio to be notified of changes to GPIO pins &c.

guymcswain, 2021-08-16:
You are correct @felixkasza

Skip to end for the answer.
`<rant>`
I should have been more accommodating in my reply and I would have had the op prefaced this issue with something like ‘hey I’m new to JavaScript (and pigpio) and I’m looking for help as to why my script is not working’.  Instead it comes across ‘as this lib is broken,’.  `</rant>`

Answer:  replace the for loop with
`let i = 0, loop = setInterval(console.log, 1000, “a “+ i++);`

`setTimeout( clearTimeout, 15000, loop);`