joan2937/pigpio #487: Not issue but Doubt about Steady value in set_glitch_filter

fiurato74, 2021-09-03:
Hello,
I am trying to use this wonderful library but I have a doubt.
In the set_glitch_filter function, the "steady" value is at most 300000 microsendonds, is there a reason why it was chosen or is it an arbitrary value?

Thank you very much.

guymcswain, 2021-09-03:
Is this value obtained from the source code or is it your observation of the behavior?

joan2937, 2021-09-03:
There are a lot of such constants.

http://abyz.me.uk/rpi/pigpio/cif.html#steady

The rationale behind selecting values was to give something reasonable as a limit.  The limit itself mainly exists as something to check against for the purpose of detecting programming errors.

fiurato74, 2021-09-04:
I know that there are many of these constants and my question was asked to understand if it was a necessary "limit" or a value that can be altered at will (obviously by changing the constant).

The purpose, of course, is to report only after the value has remained stable for a time of "steady" microseconds to avoid receiving "glitches" during the input; however, in my case, this time is required to be greater (in the order of seconds) and having already this useful function I think it is superfluous to implement another one, this only if the value of 300000 has not been chosen for "necessity".

guymcswain, 2021-12-28:
We are not changing the library to suit a specific need.  You can easily recompile the library with a constant of your choice.

closed 2021-12-28 by guymcswain