joan2937/pigpio #488: Wiegand reader not 'responding' to RF card scans

fabkoh, 2021-09-10:
I am currently testing the Python version of the Wiegand decoder. So far, I have tested with 2 different off-the-shelve Wiegand card readers with both having the same behaviour. Both readers come with keypads and when the keys are pressed, the output from the Python script behaves as expected (e.g. pressing 6 on the keypad returns "bits=4 value=6"). However, when I flash an RF card on the reader, nothing happens.

I believe the decoder is already working since the keypresses are returning the correct values, I have also ran piscope to look at the signal jumps each time a key is pressed. I've look at the forums to find out what's missing but haven't seen anyone with a similar issue to mine. Am I missing something obvious here? Any help would be greatly appreciated.

My code: (the only things I've changed are the pin numbers and timeout values)

`#!/usr/bin/env python

import pigpio

class decoder:

   def __init__(self, pi, gpio_0, gpio_1, callback, bit_timeout=5):

      self.pi = pi
      self.gpio_0 = gpio_0
      self.gpio_1 = gpio_1

      self.callback = callback

      self.bit_timeout = bit_timeout

      self.in_code = False

      self.pi.set_mode(gpio_0, pigpio.INPUT)
      self.pi.set_mode(gpio_1, pigpio.INPUT)

      self.pi.set_pull_up_down(gpio_0, pigpio.PUD_UP)
      self.pi.set_pull_up_down(gpio_1, pigpio.PUD_UP)

      self.cb_0 = self.pi.callback(gpio_0, pigpio.FALLING_EDGE, self._cb)
      self.cb_1 = self.pi.callback(gpio_1, pigpio.FALLING_EDGE, self._cb)

   def _cb(self, gpio, level, tick):

      if level < pigpio.TIMEOUT:

         if self.in_code == False:
            self.bits = 1
            self.num = 0

            self.in_code = True
            self.code_timeout = 0
            self.pi.set_watchdog(self.gpio_0, self.bit_timeout)
            self.pi.set_watchdog(self.gpio_1, self.bit_timeout)
         else:
            self.bits += 1
            self.num = self.num << 1

         if gpio == self.gpio_0:
            self.code_timeout = self.code_timeout & 2 # clear gpio 0 timeout
         else:
            self.code_timeout = self.code_timeout & 1 # clear gpio 1 timeout
            self.num = self.num | 1

      else:

         if self.in_code:

            if gpio == self.gpio_0:
               self.code_timeout = self.code_timeout | 1 # timeout gpio 0
            else:
               self.code_timeout = self.code_timeout | 2 # timeout gpio 1

            if self.code_timeout == 3: # both gpios timed out
               self.pi.set_watchdog(self.gpio_0, 0)
               self.pi.set_watchdog(self.gpio_1, 0)
               self.in_code = False
               self.callback(self.bits, self.num)

   def cancel(self):

      self.cb_0.cancel()
      self.cb_1.cancel()

if __name__ == "__main__":

   import time

   import pigpio

   import wiegand

   def callback(bits, value):
      print("bits={} value={}".format(bits, value))

   pi = pigpio.pi()

   w = wiegand.decoder(pi, 2, 3, callback)

   while True:
       time.sleep(60)

   w.cancel()

   pi.stop()`

joan2937, 2021-09-11:
I do not understand the problem.

The Wiegand software is passive, it simply listens for transitions on the two data lines.

Are you saying those transitions are present and are not being processed?  I.e. they are seen by piscope but not by the Wiegand software?

If so please provide a piscope capture of the Wiegand reading.  If not so this has nothing to do with pigpio.

fabkoh, 2021-09-11:
Thanks @joan2937 for the response. You're right, the piscope does not capture any data from the D0/1 lines when an RF card is scanned on the reader. However, it does transmit data when a key is pressed on the keypad. At first, I thought this was a reader specific issue, so I got another reader which was a different brand to test and I got the same behaviour. (Keypad presses transmit data over D0/1, but nothing happens when I flash an RF card on the reader).

So I was wondering if there was anything I'm missing in my code (or any step) that is required to 'activate' the antennas on the readers?

joan2937, 2021-09-11:
I have no idea how the RF card and reader you are using works.

This is not an issue with pigpio, it will only be an issue with pigpio if correct Wiegand data is sent but is incorrectly processed by pigpio.

closed 2021-09-11 by joan2937

fabkoh, 2021-09-13:
@joan2937 I found the problem. Just to close this out, the issue was basic but wanted to share in case anyone comes across it in future.  I was scanning MIFARE (13.56Hmz) cards on a reader which was for EM (125Khz), hence nothing was happening when I scanned the cards. Had to look for some EM cards to test and it works fine now. Thanks for the responses.