joan2937/pigpio #489: Raspberry pi SPI3,4,5,6 mode how to set

dipenpatel235, 2021-09-18:
hello @@joan2937 i have seen your below code for enable spi 3 ,4 , 5 ,6  but i don't get any idea how to enable it.
https://github.com/joan2937/pigpio/issues/278#issuecomment-719518470

below is my sample code in that i am able to set spi1 with mode 2 using below syntax
```
    h = pi.spi_open(0, 1000000, 258)
```

but i don't know what need to do for spi3,4,5,6
please help me what need to change 

thanks

joan2937, 2021-09-18:
You will have to use the linked fork of pigpio.  As far as I know it has not been incorporated into the pigpio master branch.

dipenpatel235, 2021-09-19:
It's ok.  I will used that. But which types of spi open i need to write it
on code ?

Can you please give spi line to open it.

guymcswain, 2021-12-28:
> but i don't know what need to do for spi3,4,5,6

pigpio does not support SPI peripherals 3, 4, 5, 6 (only available on RPi4).

closed 2021-12-28 by guymcswain