joan2937/pigpio #490: Python - I2C slave issue - Use of bsc_xfer and bsc_i2c

matbs24, 2021-09-30:
I would use a Rpi 3B+ as I2C slave. I had the following issue: Each reply to master contains the previous byte put in the FIFO.  Slave is always a byte behind respect to what is supposed to send. 

I did a quick test to explain this:
I2C communication
Master: Arduino uno sending incremental byte every second. Starting from value 1

```
#include <Wire.h>

void setup()
{
   Wire.begin(); // join i2c bus as master
   Serial.begin(9600);  // start serial for output
   while(Serial.available() == 0) {
    }
   int mydata = Serial.read();
}

char strOut[17],strIn[17];

int x = 0;

   
void loop()
{   
  while (x < 20){
     if (++x > 9999999) x=0;
     sprintf(strOut, "Sending %d \n", x);
     Serial.print(strOut);
     
     Wire.beginTransmission(24); // transmit to device #24 = 0x18
     Wire.write(x);           // sends 16 bytes
     Wire.endTransmission();    // stop transmitting
     
     Wire.requestFrom(24, 1);    // request 6 bytes from slave device #8
  
     while (Wire.available()) { // slave may send less than requested
        char c = Wire.read(); // receive a byte as character
        sprintf(strIn, "Received %d \n", c);
        Serial.print(strIn);
     }
     delay(1000);
  }
  
}
```
Slave:Rpi 3B+ running pigpio deamon at 1 us. Acting as I2C slave. When one byte read request arrived replay back with the same byte

```
import time
import pigpio

SDA=18
SCL=19

_slave_address = 0x18

def i2c(id, tick):
   global pi

   s, b, d = pi.bsc_i2c(_slave_address)
   global stop
   if b:
        data = int.from_bytes(d,"big")
        print(data)
        print(pi.bsc_i2c(_slave_address,[data]))
        stop = stop+1
        
if __name__ == '__main__':

    pi = pigpio.pi()

    stop = 0

    if not pi.connected:
        exit()

    # Add pull-ups in case external pull-ups haven't been added

    pi.set_pull_up_down(SDA, pigpio.PUD_UP)
    pi.set_pull_up_down(SCL, pigpio.PUD_UP)

    # Respond to BSC slave activity

    e = pi.event_callback(pigpio.EVENT_BSC, i2c)

    pi.bsc_i2c(_slave_address) # Configure BSC as I2C slave

    time.sleep(25)

    e.cancel()

    pi.bsc_i2c(0) # Disable BSC peripheral

    pi.stop()
``
![screenmsa](https://user-images.githubusercontent.com/32978885/135467740-bcc3495b-c98a-438d-a46f-2461a828e57b.jpeg)
`

Furthermore I tested by setting up and down a GPIO as soon as I entered the i2c callback and this is the timing
![Capture](https://user-images.githubusercontent.com/32978885/135467962-3f1132c3-6ab7-411e-91d9-bf0959171d6d.JPG)
g

matbs24, 2021-09-30:
[image]

matbs24, 2021-09-30:
Furthermore I made a signal acquisition and I add a set of GPIO pin when entering the callback and this is the timing that partially explain the issue.
[image]

guymcswain, 2021-10-14:
From your arduino code it looks like your loop is: write, read, delay.  You should delay after write to allow time for Python client to service the BSC.