joan2937/pigpio #492: Importing the PiGPio to the Pi Pico on micropython on MacOS

jos149, 2021-10-03:
Dear reader,

I want to control a LED strip with my pi pico.
A ran into this library and succesfully tested it with my pi 4. Does wat it should.
For the next step I want to write a python code and run this on my pi pico.
The problem I run into is: having downloaded the repository to my macbook and trying to run `make` gives me:

`pigpio-master % make
gcc -O3 -Wall -pthread -fpic -c -o pigpio.o pigpio.c
pigpio.c:59:10: fatal error: 'sys/sysmacros.h' file not found
#include <sys/sysmacros.h>
         ^~~~~~~~~~~~~~~~~
1 error generated.
make: *** [pigpio.o] Error 1`

which raises the question: How would I install this library to my mac?
What I want to do is to run a python code telling:
`from machine import pigpio`
To be able to run the commands on my pi pico.

Any help would be appreciated.

joan2937, 2021-10-03:
You can't run pigpio on the pico.  It only runs on Raspberry Pis.

I understand your confusion, just accept that the Raspberry Pico is not a Raspberry Pi.

I have written a simple daemon for the pico, but it is not compatible with micropython.

See http://abyz.me.uk/picod/ or https://github.com/joan2937/picod

jos149, 2021-10-03:
Thanks for your quick reply.

Then I am still wondering: is it possible to import the pigpio library on macOS using the thonny software? How would you do it?

Same for the pico software:
I can download the library but will I get a similar error running 'make'?

closed 2021-12-28 by guymcswain

kanth1, 2022-06-28:
when i execute below command on my mac

`user@XYZuser pigpio-master % make`
i get the below error: 
pigpio.c:59:10: fatal error: 'sys/sysmacros.h' file not found
#include <sys/sysmacros.h>
         ^~~~~~~~~~~~~~~~~