joan2937/pigpio #498: Delay between SPI calls

Yinameah, 2021-11-04:
Hello,

Before all, thanks to the developpers for the amazing work with pigpio. I've used it on a few project already, very cool.

I'm used to python programming, but I'm not an expert in low-level stuff, so sorry if the question is annoying or anything.

I'm currently trying to drive a device via a SPI link up to 24MHz. Various short messages (5-8 bytes) have to be exchanged with the slave chip. I'm using the python interface to pigpio, locally on a Raspberry Pi4.

The problem I'm having is that I end up with long delay between the SPI calls.

[image]
In this screenshot, we can see the 5 bytes being transmitted really fast, and a big wait after that.

The code is really simple (just for test) :
```py
pi = pigpio.pi()
spi_handle = pi.spi_open(0, 12_000_000, 4)
while True:
    count, rx_data = pi.spi_xfer(spi_handle, b'\x00\x01\xAC\x4A\x12')
```
So, I'm aware that we are in python and we won't reach C level speeds. But still, it feels weird to me that's the best achievable speed.

Is it expected performance ? Isn't it some kind of configuration more suited to such short messages ? The delay between the two spi transaction is 50x bigger than actual transmission. It just feels weird. Is the bottleneck on the socket level ? 

Thank you in advance to anyone that could enlighten me with some better understanding of the phenomenon.

joan2937, 2021-11-05:
I'm afraid that delay is to be expected.  The pigpio Python module talks to the pigpio daemon via socket calls.  Each call to `spi_xfer` means a network message to and from the daemon.

If you are running on the local Pi I suggest you use the Python spidev module for SPI.  This will be considerably faster as spidev calls will have much less overhead.

Yinameah, 2021-11-06:
Ok, so, it is an expected delay from the network interface. I was afraid of this, but thank you very much for confirming it.

And thanks for pointing to spidev.

closed 2021-11-06 by Yinameah