joan2937/pigpio #502: pigpiod reach more then 8~12 load average

Jernel-dotcom, 2021-11-20:
pigpiod reach 8~12 cpu load on Raspberry Pi 4 Model B (aarch64) (4 cpu)

root@DietPi:/dev# cat /etc/*release*
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian

root@DietPi:/dev# uname -r
5.10.63-v8+

top - 15:46:03 up 31 min,  2 users,  load average: 13.02, 11.07, 6.64
Tasks: 122 total,   1 running, 121 sleeping,   0 stopped,   0 zombie
%Cpu(s):  4.3 us,  1.4 sy,  0.0 ni, 94.2 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
MiB Mem :   3844.9 total,   3316.7 free,    276.3 used,    252.0 buff/cache
MiB Swap:      0.0 total,      0.0 free,      0.0 used.   3470.2 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                                      
  288 root      20   0  337540   1760   1540 S   6.7   0.0   1:48.67 pigpiod                                                                                                      
13211 root      20   0   10640   3084   2632 R   6.7   0.1   0:00.01 top                                                                                                          
    1 root      20   0  166088   9428   7096 S   0.0   0.2   0:11.64 systemd                                                                                                      
    2 root      20   0       0      0      0 S   0.0   0.0   0:00.01 kthreadd  

no results for -
cat /dev/pigerr
cat /deb/pigpio

didn't help -
pigs csi 0xf0 mils 2000 csi 0

on this OS only nextcoloud app is running

guymcswain, 2021-11-20:
6.7% seems reasonable.  You can’t turn off alerts unless you restart the daemon with appropriate flags set.

Jernel-dotcom, 2021-11-25:
Sorry I didn't get it, can you please clarify?

joan2937, 2021-11-26:
pigpio is sampling the GPIO in the background 200 thousand times per second.  It takes about 7% of a core to process those samples.  You can sort of stop the processing but in that case you might as well be using a different library.

closed 2021-12-28 by guymcswain

thecoop, 2025-07-16:
@joan2937 Is it possible to add a configuration setting to rate-limit the GPIO samples, so it could be a bit lighter on the CPU?