joan2937/pigpio #505: Getting older versions?

Joey-1970, 2021-12-20:
Hello!

Can I get somewhere a older version? Something with I2C do not work. I'm not sure where is the error. To take sure that this is not a problem with version 79 I want to check with an older version...

Joachim

guymcswain, 2021-12-20:
Yes.  Click on 'code' tab then select 'tags' and you will find earlier versions.   Let me know if you are still seeing problems.

guymcswain, 2021-12-20:
'code' -> 'master' -> 'tags'

joan2937, 2021-12-20:
Some versions prior to git are also available from https://abyz.me.uk/rpi/pigpio/VERSIONS/

```
pigpio-V3.tar
pigpio-V4.tar
pigpio-V5.tar
pigpio-V6.tar
pigpio-V7.tar
pigpio-V8.tar
pigpio-V9.tar
pigpio-V10.tar
pigpio-V11.tar
pigpio-V12.tar
pigpio-V13.tar
pigpio-V14.tar
pigpio-V15.tar
pigpio-V16.tar
pigpio-V17.tar
pigpio-V18.tar
pigpio-V19.tar
pigpio-V20.tar
pigpio-V21.tar
pigpio-V22.tar
pigpio-V23.tar
pigpio-V24.tar
pigpio-V25.tar
pigpio-V26.tar
pigpio-V27.tar
pigpio-V28.tar
pigpio-V29.tar
pigpio-V30.tar
pigpio-V31.tar
pigpio-V32.tar
pigpio-V33.tar
pigpio-V34.tar
pigpio-V35.tar
pigpio-V36.tar
pigpio-V37.tar
pigpio-V38.tar
pigpio-V39.tar
pigpio-V40.tar
pigpio-V41.tar
pigpio-V42.tar
pigpio-V43.tar
pigpio-V44.tar
pigpio-V45.tar
pigpio-V46.tar
pigpio-V47.tar
pigpio-V48.tar
pigpio-V49.tar
pigpio-V50.tar
pigpio-V51.tar
pigpio-V52.tar
pigpio-V53.tar
pigpio-V54.tar
pigpio-V55.tar
pigpio-V56.tar
pigpio-V57.tar
pigpio-V58.tar
pigpio-V59.tar
pigpio-V60.tar
pigpio-V61.tar
pigpio-V62.tar
pigpio-V63.tar
pigpio-V64.tar
```

Joey-1970, 2021-12-21:
Hello Guy,
Hello Joan,

many thanks for your support. I turn back to V68 - the version I'm sure it works - but the failure is still there. I think that is **not** a issue of pigpio.
I send commands over a socket to a other raspberry pi, that works fins for a very long time, but now sometimes it work, but mostly not. I get back errors from pigpio like "PI_I2C_READ_FAILED" or "PI_I2C_WRITE_FAILED". So I find out that I give to less time in my application to operate with pigpio, now I give more and it look like ist works...
Now I must look why it takes more time then before...

Joachim

closed 2021-12-28 by guymcswain