joan2937/pigpio #506: Cannot read DHT sensor when pigpiod is running

Compizfox, 2021-12-23:
I have a Raspberry Pi 1B+ with a AM2302/DHT22 sensor connected to pin (BCM)`2`. Pins `17`, `22`, and `27` drive a LED strip using PWM, controlled by Home Assistant via pigpiod.

I have a Python script which periodically reads the DHT sensor using the `Adafruit_DHT` library.

My issue is that when pigpiod is running, I can no longer read the DHT sensor. When I stop the pigpiod service, I can read it again without problems.

Any ideas? Pigpiod shouldn't affect unrelated pins, right?

guymcswain, 2021-12-23:
> when pigpiod is running, I can no longer read the DHT sensor.

It is not clear what you mean by this because the Python pigpio client requires pigpiod to be running.  Perhaps you meant to say: 'When the HA service (which, presumably, also communicates with pigpiod) is running, I can no longer read the DHT sensor.'

Please clarify.  Thanks.

Compizfox, 2021-12-23:
I'm not using the Python pigpio client. I'm using the `Adafruit_DHT` Python library independently of pigpiod to read the DHT sensor.

When pigpiod is running (no matter if HA is connected to it or not), reading the DHT sensor fails.

guymcswain, 2021-12-23:
Ah, thanks for the clarification.  I don't know the Adafruit lib but it is unlikely compatible with pigpio.  There are examples of using pigpio to read DHT22 sensor that would avoid a clash between different libraries.  However, I don't know exactly what the HA client is doing so there may still be an underlying issue.

Compizfox, 2021-12-23:
Ah, so this is normal (while pigpiod is running, it's not possible to access GPIO pins via other means)? That wasn't clear to me. I thought that still would be possible, as long as you don't try to manipulate the same pins at the same time.

guymcswain, 2021-12-23:
Yes.  This is because pigpio does direct register level access to the peripheral hardware.  As long as all IO is done through pigpio then different pins can be independently manipulated.

Compizfox, 2021-12-23:
I see. Thanks for clarifying. The idea of that being the case crossed my mind, but I couldn't find any info on it.

I modified my Python script to use `pigpio-dht` instead of `Adafruit_DHT`, and that appears to be working!

closed 2021-12-23 by Compizfox