joan2937/pigpio #507: RPI4 - Problems using SPI

TMRh20, 2022-01-08:
Im using a RPi4B and having problems porting my library ( https://github.com/nRF24/RF24 ) over to pigpio.

The SPI functionality seems to work when doing multi-byte transfers, but I'm guessing there is a problem with single-byte transfers when using the spiXfer(); function. I don't have a logic analyzer to see exactly what is going on, but when I try to call spiXfer with multi-byte transfers it works fine. If I change the code to transmit/receive one byte at a time, it fails.

My code works fine on earlier model RPis (Raspberry Pi Model B (Rev 2.0, 512Mb)) but not on the RPI4.

I'm wondering a: is RPi4B fully supported in pigpio? b: Is SPI known to work on RPI4B?

The code in question is at https://github.com/nRF24/RF24/blob/pigpio_driver/utility/pigpio/spi.cpp

guymcswain, 2022-01-08:
> I'm wondering a: is RPi4B fully supported in pigpio?

Some added peripherals are missing.  For example Pi4 has 6 SPI peripherals but pigpio only supports the first two which are back compatible with earlier versions of rpi.

> b: Is SPI known to work on RPI4B?

Yes, on pigpio versions that support Pi4.  What version are you running?

Can you share the code where you are calling the api and also show what the return values are.

TMRh20, 2022-01-08:
I first had the package installed from APT, but then tried compiling directly from github source. Version currently reported is 79

The API is called in the following files: 
https://github.com/nRF24/RF24/blob/pigpio_driver/utility/pigpio/gpio.cpp
https://github.com/nRF24/RF24/blob/pigpio_driver/utility/pigpio/spi.cpp
https://github.com/nRF24/RF24/blob/pigpio_driver/utility/pigpio/interrupt.cpp

I've checked all the return values, and they are inline with what the documentation says they should be:
spiXfer is returning the correct number of bytes
spiOpen returns a valid handle
gpioSetMode returns 0
gpioWrite returns 0 

I also tested basic GPIO functionality with a LED, and it is toggling on and off.
I am getting valid SPI data back most of the time, so I'm not sure exactly where the problem is.

TMRh20, 2022-01-09:
FYI I had somebody else test my driver implementation and they got it working fine. I have it working mostly, but I think something was/is wonky with my RPI4 or user error, not the pigpio driver.
Thanks for the info and the software!

closed 2022-01-09 by TMRh20

2bndy5, 2022-01-09:
> I first had the package installed from APT, but then tried compiling directly from github source.

Is it possible that the RF24 build was linking to the pigpio installed from apt instead of from src? Did you purge pigpio from `/usr/lib` & `/usr/include`? Installing from src puts pigpio in /usr/local/lib & /usr/local/include. I think this is a common gotcha with apt vs src (including deb/rpm pkgs) installs, so I figured I'd ask.

TMRh20, 2022-01-09:
It is possible, I didn't purge `/usr/lib` or `/usr/include`. I was making a lot of changes trying to get it working, so its looking like some sort of user error.