joan2937/pigpio #511: bsc_xfer for RPI4B (BCM2711) has different register structures

Quest2GM, 2022-01-10:
Hi,

I noticed that for bsc_xfer (http://abyz.me.uk/rpi/pigpio/python.html#bsc_xfer), the structure of the control and status registers in the documentation are referenced from the BCM2835, as seen on pg.164 - 166 here: https://www.raspberrypi.org/app/uploads/2012/02/BCM2835-ARM-Peripherals.pdf

But, for the RPI4B which is based on the BCM2711, the control and status registers and have a completely different structure, as seen from pg. 24-27 here: https://datasheets.raspberrypi.com/bcm2711/bcm2711-peripherals.pdf.

Are these updated register structures reflected in pigpio for the BCM2711? If so, it would be nice to see that in the documentation, which only reflects the BCM2835. 

The reason I ask is because the status register on a RPi4B setup doesn't appear to return useful information. I keep getting a status of "18" no matter what state the Pi is in in its I2C operation, when running the following command: ```python status, count, data = pi.bsc_i2c(I2C_ADDR)```. While the status register doesn't work however, the data is correctly read.

I don't know if I am doing something wrong or if pigpio has not been completely supported for the BCM2711.

Thanks for any help.

guymcswain, 2022-01-11:
Thanks for pointing this out.

I will need to find some time to read through the latest version of the documentation to see what may have changed.  Just so you know, a lot of the early work to integrate the BC2711 was done initially without any released documentation so as soon as the first documents were released the support was 'finalized'.  In addition, the BSC was never fully documented in the BC2835 specification.  Specifically, the support for BSC in SPI peripheral mode was implemented from information provided by someone who reverse engineered the functionality and provided documentation on the raspberry pi forum.  Given the history, I will not be surprised if gaps exist in the current implementation of the BSC peripheral.

If you are inclined to roll up your sleeves and help, I would be grateful.  Otherwise, as I said, I will need to find the time. 

> I keep getting a status of "18" no matter what state the Pi is in, when the following command is run: status, count, data = pi.bsc_i2c(I2C_ADDR).

If you could provide a simple test case that demonstrates what you believe is incorrect behavior along with an explanation as to why, it would help me prioritize this issue.

Quest2GM, 2022-01-12:
Thanks @guymcswain for your response.

As it turns out, I think there was just an issue with my code that was causing it to only return one status. So after the fix, I am now getting different status numbers at different states.

As far as the documentation on the BCM2835 vs that of the BCM2711, I realized that there is an important difference; the 2835 has a specific section for "SPI/BSC Slave" (Section 11), whereas the 2711 does not mention anything about the BSC slave (it seems that pg. 24-27 which I had referred to in my original post seem to talk about registers in a general sense and not specific to the BSC slave). Maybe there is a gap in the 2711 documentation, or maybe it implies that the BSC slave stuff did not change from the 2835. Perhaps mentioning this in the pigpio documentation on ```bsc_xfer``` would remove some confusion for future users.

Anyhow, thanks for your help!