joan2937/pigpio #512: Multiple square waveform with different frequencies?

openedhardware, 2022-01-13:
I'd like to generate a couple of square waveforms with different frequencies on different pins(duty cycle is 50%).
e.g.

- 20kHz on GPIO26
- 30kHz on GPIO19

I am using the following code to generate, but the waveform on GPIO19 isn't 30kHz at all:
```python
import time

import pigpio

pi = pigpio.pi()

conf = {        # pin: Hz
    26: 20000,
    19: 30000
}

for pin, freq in conf.items():
    pi.set_mode(pin, pigpio.OUTPUT)
    pulse_width = int(1. / freq * 1000000) // 2  # us
    actual_frequency = int(1000000 / pulse_width / 2)
    square = [
        pigpio.pulse(1 << pin, 0, pulse_width),
        pigpio.pulse(0, 1 << pin, pulse_width)
    ]
    pi.wave_add_generic(square)
    print(f"Waveform on GPIO{pin}, {freq}(actually {actual_frequency})Hz")

wave_id = pi.wave_create()
if wave_id >= 0:
    print(f"Wave ID: {wave_id}")
    pi.wave_send_repeat(wave_id)

while True:
    try:
        time.sleep(.1)
    except KeyboardInterrupt:
        break

if wave_id >= 0:
    pi.wave_tx_stop()
    pi.wave_delete(wave_id)

pi.stop()
```

Result:
```
Waveform on GPIO26, 20000(actually 20000)Hz
Waveform on GPIO19, 30000(actually 31250)Hz
Wave ID: 2
```
Waveform on GPIO19(should be **31.25kHz**, but 20kHz actually):

[image]

Is it impossible to generate different waveforms on multiple pins?

guymcswain, 2022-01-13:
Yes, if you repeat with a period that is an integral multiple of the two of the two periods you want to generate.  In this case try placing two cycles of 20KHz and three cycles of 30KHz within a wave repeating at 10 KHz.

openedhardware, 2022-01-13:
Thanks for your reply, @guymcswain 

> Yes, if you repeat with a period that is an integral multiple of the two of the two periods you want to generate. In this case try placing two cycles of 20KHz and three cycles of 30KHz within a wave repeating at 10 KHz.

Could you share a code snippet to generate the 10kHz period?

closed 2022-01-30 by guymcswain