joan2937/pigpio #513: using write() to gpio in INPUT mode causes mode is changed to OUTPUT?

AlfaBravoX, 2022-01-14:
Hi, this is just generic question. I noticed if I use write() to INPUT mode GPIN, then state is changed yes, but mode to OUT is changed too. Is this expected behaviour?

guymcswain, 2022-01-14:
> I use write() to INPUT mode GPIN, then state is changed

I will need to see your specific code example to precisely understand what your question is.  The GPIO pin level can be read even when the mode is set to output.  For example, using pigpio, you can generate a waveform on a gpio (write) while simultaneously receiving notifications every time the level changes (read).  But I'm not sure what you are asking until I see the code snippet.

guymcswain, 2022-01-14:
Answering the question in the title directly - "using write() to gpio in INPUT mode causes mode is changed to OUTPUT?" - the answer is yes this is expected.

AlfaBravoX, 2022-01-14:
This is the code snippet 

`import pigpio

_gpin_a = 27
pi_gpio = pigpio.pi()

pi_gpio.set_glitch_filter(_gpin_a, 500)
pi_gpio.set_mode(_gpin_a, pigpio.INPUT)
pi_gpio.set_pull_up_down(_gpin_a, pigpio.PUD_UP)

print("gpin #{} mode is: {}".format(_gpin_a, pi_gpio.get_mode(_gpin_a)))
pi_gpio.write(_gpin_a, 0)
print("gpin #{} mode is: {}".format(_gpin_a, pi_gpio.get_mode(_gpin_a)))
pi_gpio.stop()`

joan2937, 2022-01-14:
If you write to a GPIO the intention is clear, you want to set its level high or low.  To do that it has to be set to output mode.

If you read a GPIO you imply nothing about the intended mode of a GPIO, it can be read regardless of the current mode.  So the mode is not altered.

AlfaBravoX, 2022-01-15:
> If you write to a GPIO the intention is clear, you want to set its level high or low. To do that it has to be set to output mode.
> 

Well not always. For example I have scenario where GPIN is used as INPUT type and most of the time it deals with connected HW switch events (and using callbacks), however same time this pin can be managed remotely from WEBSITE. So If I need to change its level high or low i need to use PUD_UP or PUD_DOWN in in order to maintain callbacks working.

guymcswain, 2022-01-30:
The `pigpio.set_mode` API is written to ensure pins otherwise used for PWM or servo do not glitch when changing to pure output mode.   It sounds to me like you have a special application/requirement.  You can create your own custom function, then call it using `pigpio.custom_1()` for example.

If you need more information please re-open this issue.

closed 2022-01-30 by guymcswain