joan2937/pigpio #514: pigpio deamon performance optimization

matbs24, 2022-01-18:
Hi,

I wanted to ask how it is possible to optimize performance of pigpio daemon and respective pigpio module used in a python script.

My specific goal is to have the best timing performance using the [SPI function](https://abyz.me.uk/rpi/pigpio/python.html#spi_xfer).
I looked at [pigpio daemon launch option](https://abyz.me.uk/rpi/pigpio/pigpiod.html) but it is still not clear to me what kind of effect they have on performance. 

Thanks

guymcswain, 2022-01-18:
> best timing performance

This is very subjective so can you please start by telling us what your requirements are and what you have tried so far.

joan2937, 2022-01-18:
Please note that pigpio uses its own SPI driver (the Linux SPI driver was not available on the Pi at the time).  Now you may get better performance by using the standard Linux SPI driver.  Which is "better" will depend on the details of your SPI transfers.

guymcswain, 2022-01-30:
If there is any issue or follow up question please re-open.

closed 2022-01-30 by guymcswain