joan2937/pigpio #518: PUD not working and massive current draw?

MichelBerg, 2022-02-10:
Hello i am new in Raspberry Pi programming, i use c++ and i want to use some GPIO pins using your library.
I have a small problem.  I wrote myself a small wrapper class for some of pigpio.h basic functions. But somehow i am not able to use the functionalitys for PULLUP/DOWN/OFF.

/**
* @brief GPIO can be set to: PULLUP / PULLDOWN / OFF
*/
typedef enum
{
    PULLUP = PI_PUD_UP,     ///< Keyword for internal Pullup
    PULLDOWN = PI_PUD_DOWN, ///< Keyword for internal Pulldown
    OFF = PI_PUD_OFF        ///< Keyword to reset internal PullUp/Down
} gpioPud_t;

void CGPIO::setPud(gpioPud_t pud)
{
    int x = gpioSetPullUpDown(m_pin, (unsigned int)pud);
    cout << "DEBUG CGPIO setPud(...) = " << x << endl;
    if (gpioSetPullUpDown(m_pin, (unsigned int)pud) != 0)
    {
        cout << "DEBUG Error@CGPIO::setPud(...)"
             << " GPIO" << m_pin << endl;
    }
}

I'll try to use GPIO23 as an INPUT with PULLUP, but while I measure it with an Voltmeter i only get 0V.

CGPIO button(23, CGPIO::INPUT);
button.setPud(CGPIO::PULLUP);
cout << "Buttonstate:" << button.digitalRead() << endl;

An other  problem is, that i want to use a GPIO as output and read it my another GPIO thats used as (normal) input.
I also made this test with an Arduino.
What suprised me is that the current draw is about 30mA for GPIO OUTPUT directly connected to GPIO INPUT. (The Raspi Datasheet says that GPIO pins can only provide 50mA.) For example the Arduino under same conditions draws only about 0.5mA. I am afraid of killing my board.

I hope somebody can help me out.

MichelBerg, 2022-02-10:
Additional Info:
I have a Raspberry Pi 4 Rev 1.4 with BCM2711.
To fix the current draw i used the functions gpioget/setPad(...) It says current can be set to 0-16ma. With an Amperemeter i get a current draw at 4, 8 ,12,16, 20, 24, 28, 31mA. (I cant find any Info why there are 4mA steps within the configuration.)

Nevertheless i used a breadboardsetup with external 10kOhm Pullup (instead of the internal PULLUP)  with a simple pushbuttonbut. Reading it with an input pin still gives me wrong results because as soon i connect the inputpin to the voltage devider it drives the output immediatly low.

MichelBerg, 2022-02-10:
Okay i'll try using the other one instead. I am aware of not shortcutting pins, but I should be able to do so while using pins as in/outputs. How am i able to read a 3.3V Logic Level? Inputs should be safe i thought.
I've got a project for my thesis and my task is to test a board designing a own testbench. Some ofe the tests consists of reading I/O Levels from the onboard controller that also runs at 3.3V.
I'll soon write more feedback after testing it on my old raspi.

guymcswain, 2022-02-10:
3.3V is safe but if you have two pins configured as outputs, one driving high the other low then you can burn out a pin.  You and I both know not to do this but mistakes are easy to make in software and then poof, there goes the pin!

Please let me know if you have good results from the rev 1.1 board.

MichelBerg, 2022-02-11:
I tested my code again on both Raspi's (1.1 & 1.4).  Luckly the mistake was made by my side.
For some reason i wrote a false parameter in my default constructor and my some of my "CGPIO" objects got the wrong parameters... I am sorry that i thought it wasn't my fault. 

But i still got a little question: Is there some functionality to set all GPIO into their initial state? I did not even know that GPIO 0-8 are standart pulled high, i just missed that in the datasheet.

But for now i am testing some of the (hard/soft) pwm functionalitys of the library.
I am very greateful using your library! :)

MichelBerg, 2022-02-11:
Just reopened if you got an answer for my little question. You can close it afterwards.

guymcswain, 2022-02-17:
> Is there some functionality to set all GPIO into their initial state?

So, as you have found out, all GPIO come up from reset in the input state with 1-8 pulled up and the others pulled down.  This is the way the hardware comes up and nothing you can do to change that.  Your application must be mindful of this fact and ignore input states until you can configure them into your desired state.

closed 2022-02-17 by guymcswain